mavlink/qgroundcontrol #12596: QGC v4.4.3 couldn't be correctly deployed on Windows

Enem-20, 2025-03-20:
I'm trying to build I'm trying to build qgc to customize it. My first build of v4.4.3 tag works correctly on linux, but I can't say the same about Windows 10. 
Now I'm creating my own link inherited from UDPLink. My link connects to the server with mavlink APM device.
fly mode is set to "Custom:0x0" (should be "Stabilize"). When I'm try to connect to my link fly mode sets to "Custom:0x0" and qgc crashes cause there is no supported classes:
`FirmwarePlugin* CompInfoParam::_anyVehicleTypeFirmwarePlugin(MAV_AUTOPILOT firmwareType)
{
    FirmwarePluginManager*  pluginMgr               = qgcApp()->toolbox()->firmwarePluginManager();
    MAV_TYPE                anySupportedVehicleType = QGCMAVLink::vehicleClassToMavType(pluginMgr->supportedVehicleClasses(QGCMAVLink::firmwareClass(firmwareType))[0]);

    return pluginMgr->firmwarePluginForAutopilot(firmwareType, anySupportedVehicleType);
}`

So I've noticed that firmwareType is equals to 3(APM), but APM plugin factory wasn't initialized as I understand.
How could I resolve this issue?

DonLakeFlyer, 2025-03-20:
If you are using custom-example as you base the custom example code is PX4 only. Change the custom .pro build file to not remove ArduPilot support.

Enem-20, 2025-03-20:
> If you are using custom-example as you base the custom example code is PX4 only. Change the custom .pro build file to not remove ArduPilot support.

I don't use custom-example and don't create plugins. I just create class MyLink : public UDPLink to receive mavlink package(telemetry) from a server and send it to 14550 port. It works properly for Linux, but not for Windows.

DonLakeFlyer, 2025-03-20:
Ok, then I don't know what's wrong. You'll have to debug it through.

Enem-20, 2025-03-20:
Tell me, why is the flight controller in custom:0x0 mode? 
I've tried to find out how APM plugin registers to  FirmwarePluginFactoryRegister, but there is nothing there that could clarify the situation. I guess the problem might be in unregistered plugins

Enem-20, 2025-03-20:
[image]

Enem-20, 2025-03-20:
[image]

Enem-20, 2025-03-20:
```
void Vehicle::_mavlinkMessageReceived(LinkInterface* link, mavlink_message_t message)
{
    qDebug() << "message id is " << message.msgid << ". for " << message.compid;
    // If the link is already running at Mavlink V2 set our max proto version to it.
    unsigned mavlinkVersion = _mavlink->getCurrentVersion();
    if (_maxProtoVersion != mavlinkVersion && mavlinkVersion >= 200) {
        _maxProtoVersion = mavlinkVersion;
        qCDebug(VehicleLog) << "_mavlinkMessageReceived Link already running Mavlink v2. Setting _maxProtoVersion" << _maxProtoVersion;
    }
...

```
This is a bug in qgroundcontrol I suppose

Vehicle receives telemetry:
message id is  0 . for  191
message id is  178 . for  1
message id is  30 . for  1
message id is  74 . for  1
message id is  178 . for  1
message id is  30 . for  1
message id is  74 . for  1
message id is  0 . for  191
message id is  0 . for  1
message id is  77 . for  1
message id is  178 . for  1
message id is  74 . for  1
message id is  30 . for  1
message id is  163 . for  1
message id is  33 . for  1
message id is  2 . for  1
message id is  193 . for  1
message id is  136 . for  1
message id is  241 . for  1
message id is  147 . for  1
message id is  1 . for  1
message id is  125 . for  1
message id is  152 . for  1
message id is  62 . for  1
message id is  42 . for  1
message id is  36 . for  1
message id is  65 . for  1
message id is  27 . for  1
message id is  29 . for  1
message id is  178 . for  1
message id is  24 . for  1
message id is  30 . for  1
message id is  74 . for  1
message id is  0 . for  191
message id is  178 . for  1
message id is  74 . for  1
message id is  30 . for  1
message id is  178 . for  1
message id is  30 . for  1
message id is  74 . for  1
message id is  178 . for  1
message id is  30 . for  1
message id is  74 . for  1
message id is  33 . for  1
message id is  2 . for  1
message id is  163 . for  1
message id is  136 . for  1
message id is  193 . for  1
message id is  241 . for  1
message id is  147 . for  1
message id is  178 . for  1
message id is  30 . for  1
message id is  74 . for  1
message id is  1 . for  1
message id is  125 . for  1
message id is  152 . for  1
message id is  62 . for  1
message id is  42 . for  1
message id is  36 . for  1
message id is  27 . for  1
message id is  29 . for  1
message id is  65 . for  1
message id is  24 . for  1
message id is  178 . for  1
message id is  30 . for  1
message id is  74 . for  1
message id is  0 . for  191
message id is  0 . for  1
message id is  178 . for  1
message id is  30 . for  1
message id is  74 . for  1
message id is  77 . for  1
message id is  163 . for  1
message id is  2 . for  1
message id is  33 . for  1
message id is  136 . for  1
message id is  193 . for  1
message id is  241 . for  1
message id is  147 . for  1
message id is  178 . for  1
message id is  30 . for  1
message id is  74 . for  1
message id is  178 . for  1
message id is  30 . for  1
message id is  74 . for  1
message id is  178 . for  1
message id is  30 . for  1
message id is  74 . for  1
message id is  163 . for  1
message id is  2 . for  1
message id is  33 . for  1
message id is  136 . for  1
message id is  193 . for  1
message id is  241 . for  1
message id is  147 . for  1
message id is  1 . for  1
message id is  125 . for  1
message id is  152 . for  1
message id is  42 . for  1
message id is  62 . for  1
message id is  36 . for  1
message id is  65 . for  1
message id is  29 . for  1
message id is  27 . for  1
message id is  24 . for  1
message id is  178 . for  1
message id is  30 . for  1
message id is  74 . for  1
message id is  0 . for  191
message id is  178 . for  1
message id is  30 . for  1
message id is  74 . for  1
message id is  30 . for  1
message id is  178 . for  1
message id is  74 . for  1
message id is  74 . for  1
message id is  178 . for  1
message id is  30 . for  1
message id is  163 . for  1
message id is  33 . for  1
message id is  2 . for  1

Enem-20, 2025-03-20:
Request for unknown firmware plugin factory 3

DonLakeFlyer, 2025-03-20:
> Request for unknown firmware plugin factory 3

If you look at the source you will find that 3 is FirmwareClassArduPilot.

> Tell me, why is the flight controller in custom:0x0 mode?

Because the flight mode names come from the firmware plugin which you don't have.

You most likely have it turned off somehow like this: https://github.com/mavlink/qgroundcontrol/blob/Stable_V4.4/custom-example/custom.pri#L26

Enem-20, 2025-03-20:
I'm using cmake. I've already tried reset local repository to v4.4.3 commit and re-clone several times. It doesn't work. 
I noticed that my windeployqt has problem with deploying this project (part of libs and qml files didn't copied). I had to do a little bit of manual work. Could it be a problem with deploying?

DonLakeFlyer, 2025-03-20:
cmake is experimental in 4.4.3. I would suggest using qmake.

Enem-20, 2025-03-20:
> cmake is experimental in 4.4.3. I would suggest using qmake.

Well, that could really have been hindrance. I'll try it out and come back with an answer!)

Enem-20, 2025-03-20:
It works properly! Thanks for your support!)