mavlink/qgroundcontrol #12597: V4.4.4

hbudney22, 2025-03-20:
The following template is for Issue/Bug reporting only. https://docs.qgroundcontrol.com/en/support/support.html#github-issues

For questions about how to use or build QGC see: http://qgroundcontrol.com/#resources

If you are using an old stable release please upgrade first and verify your issue before reporting.
----

## Expected Behavior
Please tell us what should happen as detailed as possible.
QGround Control should open.
## Current Behavior
Please describe what happens instead of the expected behavior.
It is not opening. I get this error: -------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               QGroundControl [1006]
Path:                  /Applications/QGroundControl.app/Contents/MacOS/QGroundControl
Identifier:            org.qgroundcontrol.QGroundControl
Version:               4.4.4 (4.4.4)
Code Type:             X86-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2025-03-20 17:44:09.7323 -0400
OS Version:            macOS 15.3.2 (24D81)
Report Version:        12
Bridge OS Version:     9.3 (22P3060)
Anonymous UUID:        9BA39C7D-FB63-5459-D98E-8B6CE154796C

Time Awake Since Boot: 310 seconds

System Integrity Protection: enabled

Crashed Thread:        0

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000

Termination Reason:    Namespace DYLD, Code 1 Library missing
Library not loaded: /Library/Frameworks/GStreamer.framework/Versions/1.0/lib/libz.1.dylib
Referenced from: <9FF50D22-275D-330A-B1BF-1B949D72DDE1> /Applications/QGroundControl.app/Contents/MacOS/QGroundControl
Reason: tried: '/Library/Frameworks/GStreamer.framework/Versions/1.0/lib/libz.1.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/Library/Frameworks/GStreamer.framework/Versions/1.0/lib/libz.1.dylib' (no such file), '/Library/Frameworks/GStreamer.framework/Versions/1.0/lib/libz.1.dylib' (no such file)
(terminated at launch; ignore backtrace)

Thread 0 Crashed:
0   dyld                          	    0x7ff806c9e43a __abort_with_payload + 10
1   dyld                          	    0x7ff806cb61d7 abort_with_payload_wrapper_internal + 82
2   dyld                          	    0x7ff806cb6209 abort_with_payload + 9
3   dyld                          	    0x7ff806c486e5 dyld4::halt(char const*, dyld4::StructuredError const*) + 335
4   dyld                          	    0x7ff806c45822 dyld4::prepare(dyld4::APIs&, dyld3::MachOAnalyzer const*) + 4189
5   dyld                          	    0x7ff806c44792 dyld4::start(dyld4::KernelArgs*, void*, void*)::$_0::operator()() const + 572
6   dyld                          	    0x7ff806c4427f start + 1727

Thread 0 crashed with X86 Thread State (64-bit):
  rax: 0x0000000002000209  rbx: 0x0000000000000001  rcx: 0x00007ff7bc3762b8  rdx: 0x00007ff7bc376720
  rdi: 0x0000000000000006  rsi: 0x0000000000000001  rbp: 0x00007ff7bc376300  rsp: 0x00007ff7bc3762b8
   r8: 0x00007ff7bc376320   r9: 0x0000000000000000  r10: 0x0000000000000099  r11: 0x0000000000000246
  r12: 0x0000000000000000  r13: 0x00007ff7bc376720  r14: 0x0000000000000006  r15: 0x0000000000000099
  rip: 0x00007ff806c9e43a  rfl: 0x0000000000000246  cr2: 0x0000000000000000
  
Logical CPU:     0
Error Code:      0x02000209 
Trap Number:     133

Binary Images:
       0x103b78000 -        0x105b0ffff org.qgroundcontrol.QGroundControl (4.4.4) <9ff50d22-275d-330a-b1bf-1b949d72dde1> /Applications/QGroundControl.app/Contents/MacOS/QGroundControl
    0x7ff806c3e000 -     0x7ff806cc9c7f dyld (*) <f2913392-361a-304f-b30d-486be5639e2d> /usr/lib/dyld
               0x0 - 0xffffffffffffffff ??? (*) <00000000-0000-0000-0000-000000000000> ???

[... 148 lines cut ...]
}

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Download newest macos version v4.4.4
2. mount dmg
3. try to right click to open - does not do the cancel only option
4. go to privacy and security and allow app to open
5. app immediately closes and gives that error

## System Information
When posting bug reports, include the following information
- Operating System: [e.g. iOS 12.1, Windows 8, macOS Mojave 10.14.1, Ubuntu 14]
- QGC Version: [e.g. 3.4.4]
- QGC build: [e.g., daily, stable, self-built from source, etc...]
- Flight Controller: [e.g., Pixhawk I, Pixhawk Mini, Pixhawk 2, etc.]
- Autopilot (with version): [e.g., PX4 1.8.1, Ardupilot Copter 3.6.1]

## Detailed Description
Provide further details about your issue/bug.

## Log Files and Screenshots
- [QGC Console Logs](https://docs.qgroundcontrol.com/en/settings_view/console_logging.html)
- Autopilot logs when available (post a link).
- Screenshots of QGC to help identify the current issue/bug behavior.

jwarnergithub, 2025-03-20:
I have the same problem. Running macOS 15.3.2 (24D81) on an M2.

alexivashina, 2025-03-21:
critical one for macOS

dpfrakes, 2025-03-24:
Same issue
macOS 15.3.2 on M4 Max

DonLakeFlyer, 2025-03-24:
@julianoes Something change from 4.4.3 to 4.4.4?

svlaskin, 2025-03-25:
Same issue - on both an M1 and M3 Pro, 15.3.2!

ascharlton, 2025-03-26:
Same macOS 15.3.2 M4 Max freshly wiped beforehand.

julianoes, 2025-03-27:
FYI @farhangnaderi

farhangnaderi, 2025-03-27:
For now you might go to 4.4.3 till we fix it.

farhangnaderi, 2025-03-28:
see https://github.com/mavlink/qgroundcontrol/pull/12614

closed 2025-04-16 by HTRamsey