mavlink/qgroundcontrol #12598: Error on link X. Error Connecting: Could not create port. Permission error while locking this device

Sefibrah, 2025-03-21:
As you can see from the title, this is the error I receive.

My QGC is 4.4.0 (4.4.4 has some issues rn). 

I use MacOS 15.3.1

As for my other setup, I will provide you with all the necessary screenshots, in hopes of painting my dire situation as accurately as possible:

MacOS + Cube Orange Plus + FPV Radio Telemetry components
[image]

[image]

[image]

QGC setup:
[image]

[image]

[image]

[image]

[image]

[image]

APM Planner has the same issue:
[image]

[image]

Please help... anyone... what am I doing wrong??

DonLakeFlyer, 2025-04-04:
Try turning off serial auto-connect

closed 2025-05-03 by HTRamsey

Sefibrah, 2025-05-06:
The way I solved the issue was by:

plugging the Air Module from TELEM 1 into TELEM 2... to this day I have no idea why did this even work...

[image]

ParamDeshpande, 2025-05-20:
https://ardupilot.org/copter/docs/common-telemetry-port-setup.html#overview
> "Telem1 (aka Serial1) is for MAVLink communication and supports flow control. This should be used especially for high power devices (up to 1 amp) But NOT the [RFD900 telemetry radio](https://ardupilot.org/copter/docs/common-rfd900.html#common-rfd900)"
Maybe something to do with this ?

Sefibrah, 2025-05-25:
Interesting observation @ParamDeshpande... this could likely be it 🤔

Thanks for sharing :))