mavlink/qgroundcontrol #12599: QGroundControl 4.4.4 not starting on Mac OSX 15.3.2, GStreamer issue? (QGroundControl 4.4.3 works)

mathew314, 2025-03-22:
## Expected Behavior
QGroundControl 4.4.4 starts and runs similar to QGroundControl 4.4.3

## Current Behavior
QGroundControl 4.4.3 works fine, but prompts to upgrade.
When I upgrade to QGroundControl 4.4.4 it won't start. QGC 4.4.3 continues to work.

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Download https://github.com/mavlink/qgroundcontrol/releases/download/v4.4.4/QGroundControl.dmg
2. Attempt to run by double clicking
3. App closes without displaying the main window

## System Information
When posting bug reports, include the following information
- Operating System: MacOS Sequoia 15.3.2
- Computer: Apple M4
- QGC Version: 4.4.4
- QGC build: Stable, https://github.com/mavlink/qgroundcontrol/releases/tag/v4.4.4
- Flight Controller: N/A, app won't start
- Autopilot (with version): N/A, app won't start

## Detailed Description
- (Renamed so you can install/run QGroundControl 4.4.3 and 4.4.4 side by side)
```
/QGroundControl --clear-settings --logging:full
```

Maybe relevant?
```
objc[2891]: Class GstGLCAOpenGLLayer is implemented in both /Users/mathew/Applications/QGroundControl 4.4.4.app/Contents/Frameworks/GStreamer.framework/Versions/1.0/lib/libgstgl-1.0.0.dylib (0x110691270) and /Library/Frameworks/GStreamer.framework/Versions/1.0/lib/libgstgl-1.0.0.dylib (0x11039e7b0). One of the two will be used. Which one is undefined.
objc[2891]: Class GstGLNSWindow is implemented in both /Users/mathew/Applications/QGroundControl 4.4.4.app/Contents/Frameworks/GStreamer.framework/Versions/1.0/lib/libgstgl-1.0.0.dylib (0x1106912e8) and /Library/Frameworks/GStreamer.framework/Versions/1.0/lib/libgstgl-1.0.0.dylib (0x11039e828). One of the two will be used. Which one is undefined.
objc[2891]: Class GstGLNSView is implemented in both /Users/mathew/Applications/QGroundControl 4.4.4.app/Contents/Frameworks/GStreamer.framework/Versions/1.0/lib/libgstgl-1.0.0.dylib (0x1106912c0) and /Library/Frameworks/GStreamer.framework/Versions/1.0/lib/libgstgl-1.0.0.dylib (0x11039e800). One of the two will be used. Which one is undefined.
```

Cause of the crash:
```
(QGroundControl:2891): GStreamer-WARNING **: 12:02:36.215: External plugin loader failed. This most likely means that the plugin loader helper binary was not found or could not be run. You might need to set the GST_PLUGIN_SCANNER environment variable if your setup is unusual. This should normally not be required though.

(process:2891): GLib-GObject-CRITICAL **: 12:02:36.219: g_param_spec_boxed: assertion 'G_TYPE_IS_BOXED (boxed_type)' failed

(process:2891): GLib-GObject-CRITICAL **: 12:02:36.219: validate_pspec_to_install: assertion 'G_IS_PARAM_SPEC (pspec)' failed

(process:2891): GLib-GObject-CRITICAL **: 12:02:36.219: g_object_new_is_valid_property: object class 'GstPadTemplate' has no property named 'caps'

(process:2891): GLib-GObject-CRITICAL **: 12:02:36.219: g_object_new_is_valid_property: object class 'GstPadTemplate' has no property named 'caps'

(process:2891): GStreamer-WARNING **: 12:02:36.219: Element factory metadata for 'breakmydata' has no valid long-name field

(process:2891): GLib-CRITICAL **: 12:02:36.222: g_array_append_vals: assertion 'array' failed

(process:2891): GLib-CRITICAL **: 12:02:36.222: g_hash_table_lookup: assertion 'hash_table != NULL' failed

(process:2891): GLib-CRITICAL **: 12:02:36.222: g_hash_table_insert_internal: assertion 'hash_table != NULL' failed

(process:2891): GLib-GObject-WARNING **: 12:02:36.222: ../gobject/gsignal.c:1680: parameter 1 of type '<invalid>' for signal "GstIpcPipelineSrc::forward_message" is not a value type

(process:2891): GLib-GObject-CRITICAL **: 12:02:36.222: g_object_new_is_valid_property: object class 'GstPadTemplate' has no property named 'caps'

(process:2891): GStreamer-WARNING **: 12:02:36.222: Element factory metadata for 'ipcpipelinesrc' has no valid long-name field

(process:2891): GLib-GObject-CRITICAL **: 12:02:36.222: g_object_new_is_valid_property: object class 'GstPadTemplate' has no property named 'caps'

(process:2891): GStreamer-WARNING **: 12:02:36.222: Element factory metadata for 'ipcpipelinesink' has no valid long-name field

(process:2891): GStreamer-WARNING **: 12:02:36.222: Element factory metadata for 'ipcslavepipeline' has no valid long-name field

ERROR: Caught a segmentation fault while loading plugin file:
/Users/mathew/Applications/QGroundControl 4.4.4.app/Contents/MacOS/../Frameworks/GStreamer.framework/Versions/Current/lib/gstreamer-1.0/libgstinterleave.dylib
```

## Log Files and Screenshots
- App won't start

holydust, 2025-03-26:
I'm having the same problem, QC4.4 15.3.2.

farhangnaderi, 2025-03-26:
So far what I did was to roll back to GStreamer 1.24.1 and did not fix the issue. Apparently CI is passing with this but the latest Gstreamer on macOS is 1.26. 
Need to dig dipper.

ascharlton, 2025-03-26:
Im having the same output from latest stable - app crashes hard with seg fault.  Macbook M4 Max 15.3.2 PX4 Pixhawk 6X QGC 4.4.4. Daily 5.0 launches fine and streaming works sometimes with one of my rtsp jidetech cameras (I have two: also a Siyi A8) but 90% of the time they both exhibit a continuous loop something like this:

VideoReceiverLog: Starting "rtsp://192.168.144.25:8554/main.264" , buffer 0
VideoReceiverLog: Started "rtsp://192.168.144.25:8554/main.264"
VideoReceiverLog: Already running! "rtsp://192.168.144.25:8554/main.264"
VideoReceiverLog: Starting decoding "rtsp://192.168.144.25:8554/main.264"
VideoReceiverLog: Streaming started "rtsp://192.168.144.25:8554/main.264"
VideoReceiverLog: Decoding started "rtsp://192.168.144.25:8554/main.264"
VideoReceiverLog: _onNewDecoderPad "rtsp://192.168.144.25:8554/main.264"
VideoReceiverLog: Decoding started
CameraControlLog: Requesting video stream status for: 1
VehicleLog: void Vehicle::_sendMavCommandWorker(bool, bool, const MavCmdAckHandlerInfo_t *, int, MAV_CMD, MAV_FRAME, float, float, float, float, double, double, float) command:param1-7 512 270 1 0 0 0 0 0
VehicleLog: void Vehicle::_sendMavCommandFromList(int) command:tryCount:param1-7 "MAV_CMD_REQUEST_MESSAGE" 0 270 1 0 0 0 0 0
VehicleLog: "_handleCommandAck command(MAV_CMD_REQUEST_MESSAGE) result(MAV_RESULT_ACCEPTED)"
CameraControlLog: handleVideoInfo: 1 rtsp://192.168.144.25:8554/main.264
CameraControlLog: All stream handlers done
VideoReceiverLog: Stopping "rtsp://192.168.144.25:8554/main.264"
VideoReceiverLog: Decoding stopped
VideoReceiverLog: Streaming stopped "rtsp://192.168.144.25:8554/main.264"
VideoReceiverLog: Stopped "rtsp://192.168.144.25:8554/main.264"
VideoReceiverLog: Starting "rtsp://192.168.144.25:8554/main.264" , buffer 0

I recently wiped my mac before 4.4.4 came out as I was having difficulties with streaming - I remember downgrading all the way back to GStreamer 18.6

By the way gst-launch causes a hard shutdown so silicon is not liking some gstreamer binaries.

kkouer, 2025-03-27:
I have the same problem too!

farhangnaderi, 2025-03-28:
See https://github.com/mavlink/qgroundcontrol/pull/12614

closed 2025-03-30 by HTRamsey