mavlink/qgroundcontrol #12603: Custom flight modes not shown if initialised after QGC launch

fsoest, 2025-03-26:
## Expected Behavior
PX4 allows users to define "custom modes" from a companion computer according to the specification here:
https://docs.px4.io/main/en/ros2/px4_ros2_control_interface.html [1]
The custom flight modes should appear in the flight mode list as soon as the corresponding ROS2 node is started, even if QGC is started first.

## Current Behavior
The custom flight mode is only registered if the ROS2 node is started before starting QGC. If the node is started first, QGC needs to be restarted for the mode to show up.

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Install ROS2 + PX4, MicroXRCEAgent
2. Install PX4 ROS 2 Control Interface following [1]
3. Start QGC
4. ros2 run example_mode_manual_cpp example_mode_manual

## System Information
When posting bug reports, include the following information
- Operating System: Ubuntu 22.04.5
- QGC Version: 4787edd
- QGC build: daily
- Flight Controller: px4_sitl
- Autopilot (with version): PX4 1.15.4

## Detailed Description
https://github.com/mavlink/qgroundcontrol/pull/12581 fixed the custom flight modes not working at all. Now needs a QGC restart if custom flight mode was started first.

DonLakeFlyer, 2025-04-16:
@fsoest Can you provide a log with StandardModesLog turned on for this repro case?

DonLakeFlyer, 2025-04-17:
@dakejahl Jake, maybe you could help here?

fsoest, 2025-04-21:
> [@fsoest](https://github.com/fsoest) Can you provide a log with StandardModesLog turned on for this repro case?

Will do when I am back in office.

fsoest, 2025-04-25:
[mode_first.txt](https://github.com/user-attachments/files/19910443/mode_first.txt)
[qgc_first.txt](https://github.com/user-attachments/files/19910444/qgc_first.txt)#

Here are logs with StandardModesLog for both cases of starting either QGC or the Mode first.
The mode seems to be recognised either way, but is not shown in the menu.

dakejahl, 2025-04-25:
I can look into this once I have some time but it might be a little while

DonLakeFlyer, 2025-04-25:
Ok, getting closer to understanding. The new modes do come across but the UI doesn't update. So likely some signaling problem somewhere.

julianoes, 2025-05-05:
I see this problem with QGC v.4.4.4. It never shows the custom flight modes. However, when I build the same version locally it works fine.

DonLakeFlyer, 2025-05-05:
@fsoest Created a pull with even more logging. Once a build pops out from that can you repro with both StandardModesLog and firmwarepluginlog turned on?

fsoest, 2025-05-07:
Here are the logs, I could not find firmwarepluginlog in the logging settings, do I need to look elsewhere?

[qgc-first.txt](https://github.com/user-attachments/files/20084101/qgc-first.txt)
[mode-first.txt](https://github.com/user-attachments/files/20084100/mode-first.txt)

DonLakeFlyer, 2025-05-07:
> I could not find firmwarepluginlog in the logging settings

qgc.firmwareplugin.firmwareplugin

I'll need that in the logs to figure it out

fsoest, 2025-05-12:
Thanks, found it. Here are the logs:

[mode_first.txt](https://github.com/user-attachments/files/20164516/mode_first.txt)
[qgc_first.txt](https://github.com/user-attachments/files/20164517/qgc_first.txt)

DonLakeFlyer, 2025-05-13:
So here is what is going on:
* QGC boots up and asks the firmware for the available modes
* In response to that it gets back a mode:
```
10.556 - debug: Available mode received - name: "(Mode not available)" index: 18 standard_mode: 0 advanced: false cannotBeSet: true custom_mode: 184811520 (StandardModesLog:unknown:0)
```
  * Note the `custom_mode` value of 184811520
  * It has `cannotBeSet: true` show although it's a mode it's not selectable by a user
  * No idea myself what this mode is with a name of `(Mode not available)`. Maybe an ardupilot thing?
* Then later one ROS cranks up and QGC is told there are new modes
* Now it gets a new mode:
```
    31.231 - debug: Flight Mode before: "My Manual Mode" true true virtual void PX4FirmwarePlugin::updateAvailableFlightModes(FlightModeList&) (default:unknown:0)
    31.231 - debug: Flight Mode after: "My Manual Mode" true true virtual void PX4FirmwarePlugin::updateAvailableFlightModes(FlightModeList&) (default:unknown:0)
```
* Which i think is using a duplicate custom_mode value of 184811520  so it gets ignored.

I need to update the logging to truly see if that is what is going on...

DonLakeFlyer, 2025-05-13:
I just updated #12779 with even more logging to prove/disprove my theory. When CI finishes can you generate logs again?

fsoest, 2025-05-15:
Here they are:

[mode_first.txt](https://github.com/user-attachments/files/20225955/mode_first.txt)
[qgc_first.txt](https://github.com/user-attachments/files/20225954/qgc_first.txt)

DonLakeFlyer, 2025-05-15:
@bkueng Can you help with this:
```
void StandardModes::availableModesMonitorReceived(uint8_t seq)
{
    if (_lastSeq != seq) {
        qCDebug(StandardModesLog) << "Available modes changed, re-requesting";
        _lastSeq = seq;
        request();
    }
}
```

When the vehicle tells QGC that is has new modes available (after initial connect) it seems like `_modeList` ends up just adding on to the modes from the initial connect. Shouldn't `_modeList` be cleared if modes are -re-requested? I don't see that happening anywhere?

DonLakeFlyer, 2025-05-18:
@fsoest I just updated my test pull with what I think is the fix. Can you try it? Still need to wait for final verification from Beat though to be safe.

bkueng, 2025-05-19:
> Shouldn't _modeList be cleared if modes are -re-requested?

Yes, and they were correctly replaced when I implemented it.

DonLakeFlyer, 2025-05-19:
Thanks @bkueng. I'll figure the rest out.

bkueng, 2025-05-23:
> So my thinking is to remove the code that tries to do that. Simpler, and likely correct. Comments?

I see why it was added, but I also tend towards removing it. For the standard modes protocol, there is not standardized way for translation yet.

DonLakeFlyer, 2025-05-24:
@fsoest #12910 should hopefully be the real pull that fixes this. Can you re-test?

fsoest, 2025-05-26:
The mode seems to be recognised now. However, I am getting an empty mode in the list, not sure if this is related.

[image]

[empty_mode.txt](https://github.com/user-attachments/files/20443348/empty_mode.txt)

DonLakeFlyer, 2025-05-26:
So that is coming from here:
```
    11.777 - debug: Available mode received - name: "" index: 12 standard_mode: 9 advanced: true cannotBeSet: false custom_mode: 33816576 (StandardModesLog:unknown:0)
```

There is no such thing as a standard_mode: 9 in the mavlink spec that I can find. Because of that it doesn't name it. That said there should be some fallback to naming here for this. But still getting a 9 doesn't seem right from the firmware side of things.

DonLakeFlyer, 2025-05-28:
@fsoest Do. you get the empty mode with 4.4 as well?

fsoest, 2025-05-28:
Yes, it also appears in v4.4.5