mavlink/qgroundcontrol #12612: Safety, define altitude threshold and warn user if drone crosses threshold. Base it on terrain data if provided.

jnomikos, 2025-03-27:
Good afternoon,

A client I work with would be interested in possibly contributing to open source for a feature I will be working on in the future, and I wanted to gauge if this would be a meaningful item that would be good to add to QGC. No worries if it is not.

Tell us a bit about the feature:
- What problem does it solve? 

Makes it easier for a user to fly safely given drone altitude and terrain data.

- What does it do?

**1. Altitude Safety Enhancements & Smart Warning System**
An area on the safety tab of QGC for a user to define maximum and minimum flyable altitude limits. The setting name could be "Warn on crossing altitude threshold". Periodically track drone position and if the drone crosses the altitude threshold, send a warning on QGC. Only send 1 warning per cross to avoid spamming the user. Do not send warnings if the drone is in land mode or landing during a mission. If possible, use the terrain data in this altitude.

**2. Shade the flyable altitudes between the low and high altitude levels on mission plan height view**

On the terrain bar on the bottom of plan view, shade the flyable altitude based on the maximum and minimum flyable altitude limits defined on the item before. The flyable altitude shading should move with the terrain, as terrain should be apart of the calculation

- What flight stacks must it work with? (All, PX4, ArduPilot)

All

- Relevant vehicle types? (All, multirotor, fixed-wing, VTOL, submarine, etc.)

Ideally with any vehicle type

- Are there other systems that have this feature?

Probably, but none that I know of

- What communication/integration standards does the feature rely on (e.g. MAVLink commands etc.)

It would utilize data already on the GCS provided from the vehicle through MAVLink, and terrain data.

Any additional context you can provide will make the feature easier to evaluate (e.g. mockups, detailed specification, etc.)

Drawing of what the shading could look like
[image]

DonLakeFlyer, 2025-03-28:
I think this makes sense as a good addition. 

Comments:
* The Safety settings are vehicle configuration settings as opposed to QGC application settings. The settings associated with this should be in application settings since they do not apply to a specific vehicle but to the app as a whole and any vehicle currently being monitored with it. Given that I would be the settings associated with this in the AppSetting class and in the General page.
* Doing this with terrain data while flying will be tricky from a UI/notification standpoint. The problems are:
  * You may not have internet access so querying for additional terrain data on the fly would not be possible in that case.
  * Even if you have internet if you fly to somewhere where you don't yet have terrain data there will be a delay as to when that terrain data becomes available.
  * I'd like to see some detail as to how the user would be notified of what is happening in these case.

jnomikos, 2025-03-29:
@DonLakeFlyer 

> The Safety settings are vehicle configuration settings as opposed to QGC application settings. The settings associated with this should be in application settings since they do not apply to a [...]

I agree with this.

> Doing this with terrain data while flying will be tricky from a UI/notification standpoint. The problems are:
> You may not have internet access so querying for additional terrain data on the fly [...]
    I'd like to see some detail as to how the user would be notified of what is happening in these case.

You bring up great points. I have been thinking about them recently, and I agree they are tricky situations to deal with.

- I think notifying the user through an alert if they enter an area while flying that doesn't have terrain data would probably be good enough. Something to inform the user that they are now in an area where they cannot rely on the GCS giving terrain data. If the data eventually becomes available (like if you have internet), we could have an additional alert.

- If having some warning is too flashy or annoying, we could maybe make some kind of icon at the top associated with terrain data availability. Maybe some icon which looks like a mountain and has a red or green circle next to it to represent if the vehicle is flying in an area that the GCS has terrain data on.

I am not sure how terrain data is loaded or queued up currently, but if someone is using offline maps and only flying where those offline maps are loaded, I imagine they can circumvent this potential issue. 

For people who don't necessarily want to use terrain data in their low/high altitude warning threshold, we could have a checkbox setting for that.

If you have any better ideas of how to handle these cases, feel free to share.