mavlink/qgroundcontrol #12617: Slow QGC interface on Macbook

DanieleAzz, 2025-03-28:
Hi there,
I tried to install QGC version 4.4.4 on my Macbook Air (M1 2022) with Sequoia 15.3.1, but unfortunately, it does not open.
I saw that there is a Bug issue for this opened a few days ago.

So I decided to install the 4.4.3 version. I followed the documentation and the installation went smoothly.

The problem was that once opened, it is very slow, every button I press it take at least 30 seconds to respond.

I searched on the PX4 forum and I found that in 2022 some users were having the same issue and someone suggested executing this in the terminal:

``
 cd ~/.config/org.qgroundcontrol
``

And edit and add this section at the end of the QGroundControl.ini file:

``
[LinkManager]
``
``
autoConnectZeroConf=false
``

And Magic happens! Everything works perfectly.

To see if the Zero-Conf Connection was the problem, I navigated to Select Tool>Application Settings>AutoConnect section and Enabled again the Zero-Conf. 
Doing that I had the same previous issue: Graphic slow.

## Expected Behavior
Install the v4.4.3 and work smoothly with it

## Current Behavior
Installed successfully, but GUI is very slow and laggy

## Steps to Reproduce:
1. Download from: 
[https://github.com/mavlink/qgroundcontrol/releases/download/v4.4.3/QGroundControl.dmg](https://github.com/mavlink/qgroundcontrol/releases/download/v4.4.3/QGroundControl.dmg)

2. Install following: 
[https://docs.qgroundcontrol.com/master/en/qgc-user-guide/getting_started/download_and_install.html#macOS](https://docs.qgroundcontrol.com/master/en/qgc-user-guide/getting_started/download_and_install.html#macOS)

3. Launch it and test, if it is slow, terminate it and go to the next step

4. Open the terminal and do:

``  cd ~/.config/org.qgroundcontrol ``
`` vim QGroundControl.ini `` 

 then add at the end:

``
[LinkManager]
``
``
autoConnectZeroConf=false
``

Save and launch again QGC. Everything will work as expected.

## System Information
- Operating System: Macbook Air (M1 2022) with Sequoia 15.3.1
- QGC Version: 4.4.3
- QGC build: stable

## Reference:
Thank you Vasyl_Maks 💯 
[https://discuss.px4.io/t/slow-qgroundcontrol-interface-on-macos/29113/9](https://discuss.px4.io/t/slow-qgroundcontrol-interface-on-macos/29113/9)

HTRamsey, 2025-03-28:
That's really weird, wonder if this happens in master after the threading changes. But yeah turning zeroconf autoconnection off by default is a decent solution for now

DanieleAzz, 2025-03-28:
I tested with another Mac:

-- Macbook Pro with Apple M1 Max
-- 15.3.2

And QGC v4.4.3 works perfectly without editing the file.
I disabled then the ''Zero-Conf'' from the ''Application Settings'', and it looks like the GUI is a little bit smoother.

DonLakeFlyer, 2025-04-14:
I'm a bit confused. The Issue says this is for 4.4.3. But then you say 4.4.3 works perfectly.  Is this 4.4.3 or daily?

DanieleAzz, 2025-04-24:
@DonLakeFlyer Hi! Sorry if I did not explained clearly.

The version is 4.4.3, not daily. The installation of this version is fine, the problem is on the responsiveness of the GUI.

I did 2 tests with different machines:

- TEST 1:
Using: MacBook Air (M1 2022) with Sequoia 15.3.1
QGC: 4.4.3
Result: GUI is slow to response, lag-input ❌ 
Fix:  [[LinkManager] autoConnectZeroConf=false ](https://github.com/mavlink/qgroundcontrol/issues/12617#:~:text=I%20searched%20on%20the%20PX4%20forum%20and%20I%20found%20that%20in%202022%20some%20users%20were%20having%20the%20same%20issue%20and%20someone%20suggested%20executing%20this%20in%20the%20terminal%3A)

- TEST 2:
Using: MacBook Pro ( M1 Max) with Sequoia 15.3.2
QGC: 4.4.3
Result: GUI is responsive and works as expected ✅ 

Let me know if it is clearer now. Thank you!

DonLakeFlyer, 2025-04-24:
Ok, understood. Have you tried a new daily build on the same hardware without the config hacks? The daily build is about to turn into the new 5.0 release. So we need to know if the same problem you are seeing on 4.4.3 will carry over to 5.0. If it does will we fix that problem there.

DonLakeFlyer, 2025-05-15:
Closing until I here there is a 5.0 issue here

closed 2025-05-15 by DonLakeFlyer