mavlink/qgroundcontrol #12624: No UDP connection to ExpressLRS backpack with macOS Firewall enabled

fmckeogh, 2025-03-31:
The following template is for Issue/Bug reporting only. https://docs.qgroundcontrol.com/en/support/support.html#github-issues

For questions about how to use or build QGC see: http://qgroundcontrol.com/#resources

If you are using an old stable release please upgrade first and verify your issue before reporting.
----

## Expected Behavior

MacOS firewall prompt appears on first use of QGroundControl and UDP connections work. 

At least adding QGroundControl to the firewall list manually should allow incoming connections.

## Current Behavior

No connection is possible, regardless of whether servers are added manually.

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Download and open QGroundControl
2. Connect to ELRS Backpack
3. No connection on QGC
4. Disable firewall
5. Connection works on QGC
6. Enable firewall
7. Connection breaks

## System Information
When posting bug reports, include the following information
- Operating System: 15.3.2 (24D81)
- QGC Version: 4.4.3
- QGC build: Stable
- Flight Controller: Pixhawk 1
- Autopilot (with version): ArduPilot Plane 4.5.3

## Detailed Description

I have made no networking/firewall changes on my Mac.

## Log Files and Screenshots
- [QGC Console Logs](https://docs.qgroundcontrol.com/en/settings_view/console_logging.html)
- Autopilot logs when available (post a link).
- Screenshots of QGC to help identify the current issue/bug behavior.

HTRamsey, 2025-04-07:
This isn't really something we can address in QGC

DonLakeFlyer, 2025-04-14:
@HTRamsey I'm not sure how a firewall changes things. Does this just means you can't use UDP auto-connect. But if you have a firewall you should be able to setup a manual UDP connection that will work?