mavlink/qgroundcontrol #12625: Can't build on Windows, header files not found

new-user-name, 2025-03-31:
## Expected Behavior
Included files like `#include <gst/gst.h>` should be visible

## Current Behavior
All header files in `C:\gstreamer\1.0\msvc_x86_64\include\gstreamer-1.0\gst` are not found.

## Steps to Reproduce:

git clone --recursive -j8 https://github.com/mavlink/qgroundcontrol.git
cd qgroundcontrol
git submodule update --recursive

Open project in QtCreator, try to build,

## System Information
Windows 11, QGC self build from source, master.

Strange, but `#include <gstreamer-1.0/gst/gst.h>` is working. I think something is in FindGStreamer.cmake.

MrLuckyAz, 2025-04-01:
I have also encountered the same problem, how can we solve it

new-user-name, 2025-04-01:
It seems I've found the solution with the little help from Copilot.

1) Install Msys2.
2) From inside Msys2, i.e. from it's shell MinGW64, install GTK, step 2 from here
https://www.gtk.org/docs/installations/windows
3) Add PKG_CONFIG_PATH=C:\msys64\ucrt64\lib\pkgconfig
4) Restart

Run Cmake -- no errors!..

It's strange, this snippet has to be in docs...

MrLuckyAz, 2025-04-02:
[image]   I followed you，it's still the same

kronee0516, 2025-04-02:
hey , i have some build problem too
i am now installed qt creator but 'No suitable kits found.'

[image]

new-user-name, 2025-04-02:
> I followed you，it's still the same
You are right, only Cmake part is working, I was too exited and did not check everything.

I did this, yet to no avail:

1) Changed `PKG_CONFIG_PATH=C:\msys64\ucrt64\lib\pkgconfig` to 
`PKG_CONFIG_PATH=C:\msys64\ucrt64\lib\pkgconfig; C:\gstreamer\1.0\msvc_x86_64\lib\pkgconfig`

2) In `C:\gstreamer\1.0\msvc_x86_64\lib\pkgconfig you'll find gstreamer-1.0.pc`, changed the first string to
`prefix=c:/gstreamer/1.0/msvc_x86_64`

I think something has to be done in cmake files.

new-user-name, 2025-04-02:
Here goes our third installment. You may add
`include_directories(C:/gstreamer/1.0/msvc_x86_64/include/gstreamer-1.0)`
at the bottom of findGstreamer.cmake. This clears all "header file not found" errors except one, in GstVideoReceiver.h.

MrLuckyAz, 2025-04-03:
I would like to ask, which version of QGC can control multiple UAV

new-user-name, 2025-04-03:
> That's great, bro, I've executed it correctly after add the “include_directories(C:/gstreamer/1.0/msvc_x86_64/include/gstreamer-1.0)”，and I did'n make any other mistakes as mentioned above.

Despite of our success, this way is both contingent and if I may say so stupid, like a one-time patch. Did you se the mammoth swaths of code in Cmake files?.. There is no way that creators of QGC could leave out this folder. We are doing something wrong. 

I'll try to do another try on Ubuntu. The first one was complete disaster.

HTRamsey, 2025-04-03:
pkgconfig included with gstreamer should be finding everything automatically, as that is how the CI builds are working. You can look at the Windows workflow to see how CI works, but there isn't really anything special. One thing to note is gstreamer 1.26 installs in a different location and won't work by default, so 1.22 is recommended.

new-user-name, 2025-04-03:
> One thing to note is gstreamer 1.26 installs in a different location and won't work by default, so 1.22 is recommended.
I installed to c:\gstreamer. Double-checked the log from Cmake, no other paths. And now it can't find anything from this list.

```
set(GST_PLUGINS
    gstapp
    gstcoreelements
    gstisomp4
    gstlibav
    gstmatroska
    gstmpegtsdemux
    gstopengl
    gstplayback
    gstrtp
    gstrtpmanager
    gstrtsp
    gstsdpelem
    gsttcp
    gstudp
    gstva
    gstvaapi
    gstvideoparsersbad
    gstx264
    # gstqml6
)
```

new-user-name, 2025-04-03:
> so 1.22 is recommended
`https://github.com/mavlink/qgroundcontrol/blob/master/src/VideoManager/VideoReceiver/GStreamer/README.md`

I used 1.18.x, as the link states.

HTRamsey, 2025-04-03:
Try with 1.22. 1.18 is for v4.4.

new-user-name, 2025-04-03:
> Try with 1.22. 1.18 is for v4.4.

Changed to 1.22, 
`LINK : fatal error LNK1104: cannot open file 'gstapp.lib'`

Strange, I have gstapp-1.0.lib in `C:\gstreamer\1.0\msvc_x86_64\lib`

HTRamsey, 2025-04-03:
These are new cmake files provided by gstreamer if any body in this thread has the time/ability to test this out: 

[FindGStreamer.zip](https://github.com/user-attachments/files/19594206/FindGStreamer.zip)

HTRamsey, 2025-04-04:
It's not going to be just a copy/paste thing

new-user-name, 2025-04-04:
> It's not going to be just a copy/paste thing

OK, please give me the starting point. Should I use current master, or current master with your files? Or just drop Windows and move to Linux?

new-user-name, 2025-04-04:
> It's not going to be just a copy/paste thing

In FindGstreamer.cmake from the current master 
```
set(GST_PLUGINS
    gstapp
    gstcoreelements
................................................
    # gstqml6
)
```
and further down the file
```
foreach(plugin IN LISTS GST_PLUGINS)
    pkg_check_modules(GST_PLUGIN_${plugin} QUIET IMPORTED_TARGET ${plugin})
```
it's quite clear even if I don't know Cmake syntax.

In the FindGHstreamer.cmake from your zip:
```
foreach(_gst_PLUGIN IN LISTS GSTREAMER_PLUGINS)
    # Safety valve for the custom targets above
    if ("${_gst_plugin}" IN_LIST _gst_CUSTOM_TARGETS)
        continue()
    endif()
```
and GSTREAMER_PLUGINS is not defined. I see definition in FindGstreamerMobile, but where have I to shove this file?

tnedev, 2025-04-07:
I had the same issue when I was using Github Actions to build for Windows (running Ubuntu locally and had troubles cross-compiling). 
Ended up adding extra steps to the windows.yml Actions for GStreamer. Hope this helps.

```
      - name: Install GStreamer
        uses: blinemedical/setup-gstreamer@v1
        with:
          version: ${{ env.GST_VERSION }}
          arch: x86_64
          flavour: msvc
          packages: gstreamer-1.0:gstreamer-1.0-devel:gstreamer-1.0-devtools:gstreamer-1.0-libav:gstreamer-1.0-plugins-base:gstreamer-1.0-plugins-good:gstreamer-1.0-plugins-bad

      - name: Setup GStreamer Environment
        shell: pwsh
        run: |
          # Ensure GStreamer environment variables are available
          $gstRoot = $(Get-ChildItem Env: | Where-Object { $_.Name -like "*GSTREAMER*" } | Format-Table -HideTableHeaders Name, Value | Out-String).Trim()
          echo "Found GStreamer environment variables: $gstRoot"
          
          # If not set, try to find GStreamer installation
          if (-not $gstRoot) {
            $possiblePaths = @(
              "C:\gstreamer\1.0\msvc_x86_64",
              "C:\gstreamer\1.0\x86_64",
              "$env:ProgramFiles\GStreamer\1.0\x86_64"
            )
            
            foreach ($path in $possiblePaths) {
              if (Test-Path $path) {
                echo "GSTREAMER_ROOT_X86_64=$path" >> $env:GITHUB_ENV
                echo "Found GStreamer at: $path"
                break
              }
            }
          }
```

HTRamsey, 2025-04-07:
@tnedev So what's the cause here? Installation not being in the default directory?

new-user-name, 2025-04-11:
Master builds OK after upgrading Gstreamer to 1.26.

I had to comment out this part in
C:\CPP6\qgroundcontrol\src\VideoManager\VideoReceiver\GStreamer\gstqml6gl\CMakeLists.txt

```
        if(EXISTS "${QGC_GST_QT6_PLUGIN_PATH}/RGBA_gles.frag")
            file(COPY_FILE "${CMAKE_CURRENT_BINARY_DIR}/.qsb/RGBA.frag.qsb" "${CMAKE_CURRENT_BINARY_DIR}/.qsb/RGBA.frag.qsb.external")
            qt6_add_shaders(gstqml6gl "gstqml6gl_shaders1"
                PREFIX "/org/freedesktop/gstreamer/qml6"
                OUTPUT_TARGETS gstqml6gl_shaders1
                FILES "${CMAKE_CURRENT_BINARY_DIR}/.qsb/RGBA.frag.qsb.external@glsl,100es,${QGC_GST_QT6_PLUGIN_PATH}/RGBA_gles.frag"
                OUTPUTS "RGBA.frag.qsb.external"
            )
        endif()
```

What does it mean?

caixiang2014, 2025-04-24:
@new-user-name I'm having the same issue now

new-user-name, 2025-04-24:
> [@new-user-name](https://github.com/new-user-name) I'm having the same issue now

Во you mean RGBA_gles.frag?.. I skipped it.

caixiang2014, 2025-04-25:
@new-user-name

caixiang2014, 2025-04-25:
@new-user-name 
https://github.com/mavlink/qgroundcontrol/issues/12625#issuecomment-2776834492
I use gstreamer 1.24，I'm having the same issue now
windows qt 6.6.1

caixiang2014, 2025-04-25:
`cmd.exe /C "cd . && C:\Qt\Tools\CMake_64\bin\cmake.exe -E vs_link_exe --intdir=CMakeFiles\vksdkdemo.dir --rc=C:\PROGRA~2\WI3CF2~1\10\bin\100226~1.0\x64\rc.exe --mt=C:\PROGRA~2\WI3CF2~1\10\bin\100226~1.0\x64\mt.exe --manifests  -- C:\PROGRA~1\MICROS~1\2022\COMMUN~1\VC\Tools\MSVC\1438~1.331\bin\Hostx64\x64\link.exe /nologo @CMakeFiles\vksdkdemo.rsp  /out:vksdkdemo.exe /implib:vksdkdemo.lib /pdb:vksdkdemo.pdb /version:0.0 /machine:x64 /debug /INCREMENTAL /subsystem:windows  && cd ."
LINK Pass 1: command "C:\PROGRA~1\MICROS~1\2022\COMMUN~1\VC\Tools\MSVC\1438~1.331\bin\Hostx64\x64\link.exe /nologo @CMakeFiles\vksdkdemo.rsp /out:vksdkdemo.exe /implib:vksdkdemo.lib /pdb:vksdkdemo.pdb /version:0.0 /machine:x64 /debug /INCREMENTAL /subsystem:windows /MANIFEST /MANIFESTFILE:CMakeFiles\vksdkdemo.dir/intermediate.manifest CMakeFiles\vksdkdemo.dir/manifest.res" failed (exit code 1104) with the following output:
LINK : fatal error LNK1104: 无法打开文件“gstapp.lib”
ninja: build stopped: subcommand failed.
08:06:18: 进程"C:\Qt\Tools\CMake_64\bin\cmake.exe"退出，退出代码 1 。
Error while building/deploying project vksdkdemo (kit: Desktop Qt 6.6.1 MSVC2019 64bit)
When executing step "构建"`

new-user-name, 2025-04-25:
> [@new-user-name](https://github.com/new-user-name) [#12625 (comment)](https://github.com/mavlink/qgroundcontrol/issues/12625#issuecomment-2776834492) I use gstreamer 1.24，I'm having the same issue now windows qt 6.6.1

I moved to 1.26, and gstapp.lib error vanished.

new-user-name, 2025-05-03:
As of today:
Current master + 1.22 + Windows, OK.

closed 2025-05-03 by new-user-name