mavlink/qgroundcontrol #12626: Wrong map input on joystick

federicociresola, 2025-04-01:
## Expected Behavior
1. Connect the radio controller (RadioMaster TX16S MKII) to the PC
2. Open the QGC
3. See the map of RC main inputs work (roll, pitch,yaw, throttle/thrust)

## Current Behavior
Wrong inputs mapped:
- The real pitch is mapped to the Throttle slider 
- The SA switch is mapped to the Pitch slider
- The real Thrust is not mapped in any sliders

## Steps to Reproduce:

1. Connect the RadioMaster TX16S MKII
2. Open the QGC 

## System Information

- Operating System: Ubuntu 22.04 LTS
- QGC Version: v4.3.0
- QGC build: stable
- Flight Controller: PX4 1.15.2v SITL
- Autopilot (with version): PX4 1.15.2v
- Radio Controller: RadioMaster TX16S MKII
- Firmware RC: EdgeTX v2.10.4 Centurion

## Log Files and Screenshots
[image]

federicociresola, 2025-04-11:
I have tested the PR #12639 and the new feature it works correctly but only in the master. This feature **there is not** in QGC-v4.4.4. I am looking forward the QGC-v5.0.0.

The following setup works for my **RadioMaster TX16S MK2** on Ubuntu 22.04LTS:
```bash
export SDL_GAMECONTROLLERCONFIG="0300000009120000544f000011010000,OpenTX Radiomaster TX16S Joystick,leftx:a3,lefty:a2,rightx:a0,righty:a1,platform:Linux"
```

HTRamsey, 2025-04-11:
I'll add that mapping by default for you too

PR #12639 (2025-04-08): Joystick: Allow SDL Mapping from Environment
Allows adding a custom joystick mapping or overriding an existing one via env variable
Closes #9884
Closes #12626