mavlink/qgroundcontrol #12631: PWC file missing from the stack

gianniskaldis, 2025-04-03:
Hello,
I am trying to operate the flashlights on the BlueRov2 but it gives me this message: "

[image]

How can i inlcude this file and download it without uninstalling everything?

DonLakeFlyer, 2025-04-04:
I would try contacting BlueROV on their forums. They don't look at things here much any more. Looks like the ArduSub QGC code might be out of date with latest firmwares.

gianniskaldis, 2025-04-04:
Thank you for the reply Don. I was able to figure it out by using their Cockpit BlueOs software. Had to change some autopilot configs. Thanks though...

HTRamsey, 2025-04-07:
Seems like you're good now right?

closed 2025-04-07 by HTRamsey

gianniskaldis, 2025-04-07:
Hello,
I was able to make everything work using cockpit. Now, my only issue is how
to take a screenshot. I have mapped it to my controller for the function
"camera toggle" but nothing happens and no file is donwloaded to my
downloads folder or windows downloads...
Is there something I need to change in terms of configuration? I do not
want to use windows screenshot...

Thanks

DonLakeFlyer, 2025-04-14:
You'll have to ask the BlueRobotics people about how to trigger the camera.