mavlink/qgroundcontrol #12632: The same port for local and remote ip addresses

demydd, 2025-04-03:
Hi
I have developed my udp server to receive/transmit mavlink messages to/from  destination ports specified.
The scheme is: FC <-> server <-> udp ports (GCS, etc). 
It works fine with MissionPlanner but for QCG. I have investigated the issue and could conclude that all mess is with destination port assignment procedure.

My flight controller (192.168.144.11:14552) is linked to QGC (notebook) via ethernet cable. The established link looks like as follows (Wireshark screenshot):
[image]
So QGC always uses the same port number (destination) as in source one.

If i use MissionPlanner the established connection looks like as follows: 
[image]
Thus, MP assignes **different** port number to the source one -> it works fine with the server.

if i want to run QGC via the server QGC uses local ip address **the same** as the server as well as **the same** port number -> QCS cannot connect to the source port.

Thus, it would be nice to **modify the connection procedure** to be able to resolve such an issue as it is done in MissionPlanner.

DonLakeFlyer, 2025-04-04:
Not quite sure why this't working for you. You can turn on UDPLinkLog logging to see what's going on inside QGC. You can also turn off udp auto-connect and manually configure a udp link which specifies the full server address to be what you need.

demydd, 2025-04-05:
hi @DonLakeFlyer 
thanks for your reply. Could you specify where i can find the specific log you recommend? I am quite a novice with QGC.
I have checked items here:

[image]

The log files are found here:

[image]

**2025-04-05 12-09-26 vehicle1.csv**  - contains parameter values for the session.
**2025-04-05 12-10-01.tlog**  - contains some data in two column format like this one:

(70387200, 547479390)
(7421, 503382417)
(-1358168064, 1277624352)
I have read the tlog file via python script:

`import struct`       
`import time` 
` with open(r'C:\Users\Admin\Documents\QGroundControl\Telemetry\2025-04-05 12-14-03.tlog', 'rb') as file:`  
`     while chunk := file.read(8):` 
`         data = struct.unpack('2i', chunk) ` 
`         print(data)`
`         time.sleep(1)`

I would be grateful for hints to check the info about connections and ports used.

DonLakeFlyer, 2025-04-14:
https://docs.qgroundcontrol.com/master/en/qgc-user-guide/settings_view/console_logging.html#logging-from-the-command-line

closed 2025-08-21 by DonLakeFlyer