mavlink/qgroundcontrol #12638: Can't build on Ubuntu, XKB::XKB

new-user-name, 2025-04-06:
Recent QGC master, QtCreator, Ubuntu 24. Cmake fails with

```
[cmake] CMake Error at /home/hippo/Qt/6.8.3/gcc_64/lib/cmake/Qt6Gui/Qt6GuiTargets.cmake:100 (set_target_properties):
[cmake]   The link interface of target "Qt6::GuiPrivate" contains:
[cmake] 
[cmake]     XKB::XKB
[cmake] 
[cmake]   but the target was not found.  Possible reasons include:
[cmake] 
[cmake]     * There is a typo in the target name.
[cmake]     * A find_package call is missing for an IMPORTED target.
[cmake]     * An ALIAS target is missing.
```

Tried
`sudo apt-get install build-essential libgl1-mesa-dev`
no changes.

HTRamsey, 2025-04-06:
I'm assuming you ran the dependencies installation tool?

new-user-name, 2025-04-06:
> I'm assuming you ran the dependencies installation tool?

You are right, my fault.

closed 2025-04-06 by new-user-name