mavlink/qgroundcontrol #12643: CTS held low blocking boot of SiK/Holybro Telemetry Radio

doozers-do, 2025-04-07:
## Expected Behavior
When powering on drone using battery, wireless telemetry should start transmitting and receiving

## Current Behavior
Telemetry module goes into firmware update mode

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. connect SiK/Holybro 915mHz telemetry module to telem1 - telem3 using 6-wire JST-GH-6P cable
2. power on drone via battery power
3. SiK immediately shows solid red led light

## System Information
When posting bug reports, include the following information
- Operating System: Mac OS 15.3.2 M1 Pro
- QGC Version: 4.4.3 (4.4.4 crashes on startup - there is a bug for this already)
- QGC build: stable
- Flight Controller: NXP Pixhawk 6X RT
- Autopilot (with version): px4 1.15.4

## Detailed Description
 - SiK modules firmware latest available - both modules
 - MAV_X_FLOW_CTL set to force off but not helping.
 - tried telem1 - telem3 and same issue
 - if the FC is booted via usb-c and not onboard power, telemetry module initializes without going in to firmware update mode. 
 - I had an extra JST-GH-6P and I cut the CTS line to confirm and the telemetry module works as expected when FC is booted from battery - confirms CTS is being held low on boot.

## Log Files and Screenshots
- I would be glad to provide relevant logs if you could please tell me what to collect and send.
- Sorry if this is a config that I am missing. This seems like this would be a common issue but I don't see anything relevant in the forum. The only thing I found remotely relevant is [this issue](https://github.com/PX4/PX4-Autopilot/issues/20762)
- please not cut CTS wire in "happy" picture attached

[image]
[image]

DonLakeFlyer, 2025-04-09:
Not sure what this has to do with QGC? This is all hardware side problems as far as I can tell.

doozers-do, 2025-04-09:
I'm so sorry. I realized I am in the wrong project. I will recreate in the correct place.

closed 2025-04-09 by doozers-do