mavlink/qgroundcontrol #12655: Fix "instal Qt" step for Stable_V4.4 CI

gillamkid, 2025-04-09:
The Github CI build for all platforms is currently failing for `Stable_V4.4` during the `install Qt` step. I made a tiny PR to show that the tests are failing here https://github.com/mavlink/qgroundcontrol/pull/12652

gillamkid, 2025-04-09:
A fix for the broken tool is currently in progress. Progress on the fix and a temporary way to fix can be seen here: https://github.com/jurplel/install-qt-action/pull/282

HTRamsey, 2025-04-09:
I've considered forking that action tbh

gillamkid, 2025-04-09:
looks like the issue has resolved itself. The issue may popup again in the future. See https://github.com/jurplel/install-qt-action/issues/281#issuecomment-2789853555 for explanation

closed 2025-04-09 by gillamkid