mavlink/qgroundcontrol #12657: Please unlock "All altitudes".

AndKe, 2025-04-09:
Tell us a bit about the feature:
- What problem does it solve?
After connecting to a vehicle with existing mission, I am unable to select/change altitude frame per-item because it is locked to "Relative to launch"

[image]

It would be nice to be able to change it to mixed, in order to modify the mission freely,

AndKe, 2025-04-09:
Each WP does already have it's - in this case "relative to launch" coordinate frame.
I agree that Changing the "Mission Start" element should not instantly change altitudes for the entire mission. 
It would be perfectly safe if the option to select "mixed modes" were still presented, and upon selecting it, no items were changed, but any mission item could be edited. 

The most flexible might be to allow people to bulk-change it - just enable the droplist, and upon changing selection ask if this should be applied to existing items, then and have warning about "you are changing mission coordinate frame" and "slide-to-confirm".

DonLakeFlyer, 2025-05-14:
@AndKe I think this was fixed by #12831. Can you check?

AndKe, 2025-05-14:
@DonLakeFlyer  I can confirm that the change enables a workaround: a "locked"  mission can be "unlocked" by uploading and downloading it from the vehicle - and then it is a mixed-frame mission.

However: this workaround still requires me to mess up the idea of storing the missions as .plan file.
I may make a nice survey mission, and then want to correct one approach altitude airframe, - and the only way to do so is to upload it/download it (making it no longer editable as polygons and survey items)

I'd still propose that the altitudes are possible to change in the "mission start" item, so that the mission remains as designed and can be edited as polygon etc..

AndKe, 2025-05-15:
that's true - this issue has two sides. / kicked in on two occasions.

One was that the mission from the vehicle was locked to relative altitudes, and one was unable to change to mixed modes.  (fixed now)
The other issue is that if one just make a fine-tuned and complex plan, there is no way of later changing it so that one can allow few waypoints in another reference-frame.  - or change them all to another frame.

Hence my request to not disable the altitude droplist in the "mission start" as suggested in https://github.com/mavlink/qgroundcontrol/issues/12657#issuecomment-2789197161

basically: do not lock the "all altitudes" 
when something else is selected: popup a question if the change should be applied to all existing items.
This way we can change it to "mixed" anytime, and/or use the dialog to quickly change a mission from one to another coordinate frame if wanted.

DonLakeFlyer, 2025-05-20:
So the thing that should be easily doable is to allowed a change from non-mixed to mixed. But not from one frame (including mixed) to another specific frame since that would require grovelling through the items. Doable but not right now in 5.0 lockdown.

DonLakeFlyer, 2025-05-20:
@AndKe Can you give my pull a try once it comes out of CI?

AndKe, 2025-05-20:
@DonLakeFlyer Yes, this seems like a good solution - I am always able to select mixed, and that will solve any issue. 
Thank you.