mavlink/qgroundcontrol #12665: GPS folder / project

new-user-name, 2025-04-10:
In project view, `QGC / header files`, and `QGC / source files` all folders from the list below are visible except GPS. Rescan project does not help. In file system view QtCreator says "This file is not part of any project."

```
add_subdirectory(ADSB)
add_subdirectory(AnalyzeView)
add_subdirectory(API)
add_subdirectory(AutoPilotPlugins)
add_subdirectory(Camera)
add_subdirectory(Comms)
add_subdirectory(FactSystem)
add_subdirectory(FirmwarePlugin)
# add_subdirectory(FirstRunPromptDialogs)
# add_subdirectory(FlightDisplay)
# add_subdirectory(FlightMap)
add_subdirectory(FollowMe)
add_subdirectory(Gimbal)
add_subdirectory(GPS)
add_subdirectory(Joystick)
add_subdirectory(MAVLink)
add_subdirectory(MissionManager)
# add_subdirectory(PlanView)
add_subdirectory(PositionManager)
add_subdirectory(QmlControls)
add_subdirectory(QtLocationPlugin)
add_subdirectory(Settings)
add_subdirectory(Terrain)
# add_subdirectory(UI)
add_subdirectory(Utilities)
add_subdirectory(UTMSP)
add_subdirectory(Vehicle)
add_subdirectory(VideoManager)
add_subdirectory(Viewer3D)
```

HTRamsey, 2025-04-10:
is QGC_NO_SERIAL_LINK enabled

new-user-name, 2025-04-10:
In CustomOptions.cmake I found
`option(QGC_NO_SERIAL_LINK "Disable Serial Links" ON)`,
changed to
`option(QGC_NO_SERIAL_LINK "Disable Serial Links" OFF)`

and put
`cmake_print_variables(QGC_NO_SERIAL_LINK)` in the next line.

It gives
`QGC_NO_SERIAL_LINK="ON"`
don't know why. No `CustomOverrides.cmake` file.

HTRamsey, 2025-04-10:
just clear cmake cache/config

closed 2025-04-10 by HTRamsey