mavlink/qgroundcontrol #12668: Building on arm based mac seems to be broken when following the guide

ingmarfjolla, 2025-04-10:
### Expected Behavior

When cloning and building the master branch, I should be able to build and get an app image.

### Current Behavior

At the very end of the build I run into these errors: 
```
-- Creating AppImage
/project/build/linuxdeploy-x86_64.AppImage: 1: cannot create ??@xb@8: Operation not permitted
/project/build/linuxdeploy-x86_64.AppImage: 2: @?L?LPPP: not found
/project/build/linuxdeploy-x86_64.AppImage: 1: Syntax error: Unterminated quoted string
/project/build/appimagetool-x86_64.AppImage: 1: cannot create ??@xb@8: Operation not permitted
/project/build/appimagetool-x86_64.AppImage: 2: @?L?LPPP: not found
/project/build/appimagetool-x86_64.AppImage: 1: Syntax error: Unterminated quoted string

```

and i'm not exactly sure how to resolve them. 

### Steps to Reproduce

1. Clone qgroundcontrol and follow the steps here: https://docs.qgroundcontrol.com/master/en/qgc-dev-guide/getting_started/container.html to build on mac. Specifically pay attention to using ` --platform linux/x86_64` . I'm using a fork that's some commits behind here: https://github.com/ingmarfjolla/qgroundcontrol but I don't think the instructions for building on mac have changed. 
2. After building the container, try to build qgroundcontrol and wait for the above error. 

### System Information

- OS : MacOS sequioa (arm based mac m1)
- QGC Version: master branch 
- QGC build: built from source 
- Flight controller: pixhawk 
- Ardupilot: also built from source

ingmarfjolla, 2025-04-16:
hi, just wanted to ask again and potentially see if building QGC on mac using docker is actually supposed to be supported? I'm looking at the actions, and it seems like the main target for the docker builds is for linux but maybe i'm misunderstanding how the actions work. any help is appreciated !

DonLakeFlyer, 2025-04-17:
@HTRamsey Should you be able to build a Linux X86_64 version using the docker on a Mac arm?

HTRamsey, 2025-04-18:
It was not setup with that in mind, no

ingmarfjolla, 2025-04-18:
Got it, thanks for that confirmation. Should that be removed from the documentation potentially or did i misunderstand it? 

Also some followup questions (but feel free to tell me where a better place to post these may be) : 

1. Should I be able to run the appimage in a container on mac if it was built on linux? In my container I made sure to run it like so: `CMD ["/home/user/QGroundControl-x86_64.AppImage", "--appimage-extract-and-run"]` to avoid fuse issues but that didn't work also. 

2. Is the only supported way to build on an arm mac to follow this guide: https://docs.qgroundcontrol.com/master/en/qgc-dev-guide/getting_started/#install-qt and then use the DMG?

DonLakeFlyer, 2025-04-18:
What are you trying to accomplish? I'm confused as to whether you are just trying to build a mac version of QGC that runs on an arm based mac. Or whether you are trying to cross-compile a linux version of QGC by building on a mac?

ingmarfjolla, 2025-04-18:
The latter. I wanted to write a docker compose file to run SITL with modified versions of QGC and ArduPilot which I thought I could accomplish by cross compiling the linux version of QGC.

DonLakeFlyer, 2025-04-18:
Ok, so cross-compile might be possible but now something QGC supports with it's build files.

That said you theoretically should be able to run ArduPilot SITL in a Linux VM and then run your QGC on the regular mac side of things. I do it all the time with PX4. I haven't quite figured out how to get ArduPilot SITL to cross the VM boundary correctly though. ArduPilot folks say it's possible.

closed 2025-04-18 by DonLakeFlyer