mavlink/qgroundcontrol #12670: Help with qgc android source code build~!

tms2003, 2025-04-12:
### Your Question

I am new to gqc and android, and I hope there is a document that will let me know how to build an apk file from source code. How should I start?

DonLakeFlyer, 2025-04-15:
FYI: There is very little effort going into 4.4.* stable releases at this point. We are focused on getting 5.0 out the door based on master.

DonLakeFlyer, 2025-04-16:
@HTRamsey It looks like there is a docker for building android now. But I can't find a CI workflow which verifies that it keeps working? Also in poking into the docker file it seems out of date in that it references Qt 6.6.3.

HTRamsey, 2025-04-16:
Yeah the Android docker needs to be updated/completed, I guess that's another thing that should be done ASAP before v5

DonLakeFlyer, 2025-05-15:
@HTRamsey Docker is in CI now isn't it?