mavlink/qgroundcontrol #12672: Symbol Redefinition Error during Qt 6.6.3 Build

OutSider070, 2025-04-14:
### Expected Behavior

Clean build completion when compiling with Qt 6.6.3 after merging latest upstream changes.

### Current Behavior

Build fails with symbol redefinition errors during compilation with Qt 6.6.3. The same codebase compiles successfully when using Qt 6.8.3.

### Steps to Reproduce

Build using Qt 6.6.3

### System Information

- Operating System: Windows 10
- QGC Version: Latest upstream (commit SHA: d8542d4b12a98a8b899ee2ccb113bc6d5e64679a)
- QGC build: Self-built from source
- Qt Versions: 
  - Failing: 6.6.3 (MSVC2019 64bit)
  - Working: 6.8.3 (MSVC2022 64bit)

### Log Files and Screenshots

FAILED: Release/QGroundControl.exe 
C:\Windows\system32\cmd.exe /C "cd . && C:\Qt\Tools\CMake_64\bin\cmake.exe -E vs_link_exe --intdir=CMakeFiles\QGroundControl.dir --rc=C:\PROGRA\~2\WI3CF2\~1\10\bin\100226\~1.0\x64\rc.exe --mt=C:\PROGRA\~2\WI3CF2\~1\10\bin\100226\~1.0\x64\mt.exe --manifests  -- C:\PROGRA\~1\MIB055\~1\2022\COMMUN\~1\VC\Tools\MSVC\1443\~1.348\bin\Hostx64\x64\link.exe /nologo @CMakeFiles\QGroundControl.rsp  /out:Release\QGroundControl.exe /implib:Release\lib\QGroundControl.lib /pdb:Release\QGroundControl.pdb /version:0.0 /machine:x64 /INCREMENTAL:NO /subsystem:windows && cd ."
LINK: command "C:\PROGRA\~1\MIB055\~1\2022\COMMUN\~1\VC\Tools\MSVC\1443\~1.348\bin\Hostx64\x64\link.exe /nologo @CMakeFiles\QGroundControl.rsp /out:Release\QGroundControl.exe /implib:Release\lib\QGroundControl.lib /pdb:Release\QGroundControl.pdb /version:0.0 /machine:x64 /INCREMENTAL:NO /subsystem:windows /MANIFEST:EMBED,ID=1" failed (exit code 1169) with the following output:
qrc_gamecontrollerdb_txt.cpp.obj : error LNK2005: "int __cdecl qCleanupResources_gamecontrollerdb_txt(void)" (?qCleanupResources_gamecontrollerdb_txt@@YAHXZ) 已经在 qrc_gamecontrollerdb_txt.cpp.obj 中定义
qrc_gamecontrollerdb_txt.cpp.obj : error LNK2005: "int __cdecl qInitResources_gamecontrollerdb_txt(void)" (?qInitResources_gamecontrollerdb_txt@@YAHXZ) 已经在 qrc_gamecontrollerdb_txt.cpp.obj 中定义
  正在创建库 Release\lib\QGroundControl.lib 和对象 Release\lib\QGroundControl.exp
Release\QGroundControl.exe : fatal error LNK1169: 找到一个或多个多重定义的符号
ninja: build stopped: subcommand failed.
16:56:53: 进程"C:\Qt\Tools\CMake_64\bin\cmake.exe"退出，退出代码 1 。
16:56:53: Error while building/deploying project QGroundControl (kit: Desktop Qt 6.6.3 MSVC2019 64bit)
16:56:53: When executing step "构建"

OutSider070, 2025-04-14:
qrc_gamecontrollerdb_txt.cpp.obj : error LNK2005: "int __cdecl qCleanupResources_gamecontrollerdb_txt(void)" (?qCleanupResources_gamecontrollerdb_txt@@YAHXZ) 已经在 qrc_gamecontrollerdb_txt.cpp.obj 中定义
qrc_gamecontrollerdb_txt.cpp.obj : error LNK2005: "int __cdecl qInitResources_gamecontrollerdb_txt(void)" (?qInitResources_gamecontrollerdb_txt@@YAHXZ) 已经在 qrc_gamecontrollerdb_txt.cpp.obj 中定义
  正在创建库 Release\lib\QGroundControl.lib 和对象 Release\lib\QGroundControl.exp
Release\QGroundControl.exe : fatal error LNK1169: 找到一个或多个多重定义的符号

HTRamsey, 2025-04-14:
I think this is the same issue as in #12641, qt_add_resources doesn't seem to work the same in Qt 6.6 vs 6.8.

OutSider070, 2025-04-15:
During compilation attempts with Qt 6.7.3, I encountered the same symbol redefinition errors as observed in Qt 6.6.3. Successful builds have only been achieved using Qt 6.8.3 so far.

closed 2025-05-03 by HTRamsey