mavlink/qgroundcontrol #12674: Trouble Loading QGC on 2 Mac devices.

RocketMoke, 2025-04-15:
Hello,
I am brand new to Ardupilot and QGC. I tried to load QDC on my Mac desktop running Sequoia 15.4 but have been unable as the pop up says it is incompatible with my OS?
So I try on my Mac M chip laptop and the pop up says QGC has unexpectedly Shutdown.  Any help would be greatly appreciated!
Thank you in advance,
-Rob

RocketMoke, 2025-04-15:
https://docs.qgroundcontrol.com/master/en/qgc-user-guide/getting_started/download_and_install.html#macOS

not sure what edition it is but downloaded from this link? I thought this would be the most recent?
Thank you Don!

DonLakeFlyer, 2025-04-15:
That's latest stable. Try installing a daily build and let me know if it works.

DonLakeFlyer, 2025-04-15:
@farhangnaderi, @julianoes  News to me that Mac Stable was changed to two separate arch versions. Which in turn will break the web site link. Has that change gone out into a tagged release yet? Also, why not a universal build? I think that's what daily is now.

@HTRamsey The mac daily build is a single universal isn't it?

HTRamsey, 2025-04-15:
Yeah

RocketMoke, 2025-04-16:
Guys, 
I tried both files on my desktop Mac on Sequoia 15.4 but I am still getting the same Unexpected Shutdown message.

farhangnaderi, 2025-04-16:
> Guys, 
> I tried both files on my desktop Mac on Sequoia 15.4 but I am still getting the same Unexpected Shutdown message.

Weird. This is Mac Silicon? 
I have tested it on my own setup (15.4) and no issues. 
Will take a look again.

farhangnaderi, 2025-04-16:
> What both files? Did you try a daily build?

They mean the ones here @DonLakeFlyer 

https://github.com/mavlink/qgroundcontrol/actions/runs/14205697927

DonLakeFlyer, 2025-04-16:
@RocketMoke Can. you try a daily build: https://docs.qgroundcontrol.com/master/en/qgc-user-guide/releases/daily_builds.html

RocketMoke, 2025-04-16:
"Ground Control to Major Tom!"

I am golden on both computers - the desktop and the M chip laptop!!!!!
I can not express my gratitude enough .
Will make a monetary contribution, as I am certain I will need some additional coaching.
Might even get my new Heewing T2 Cruza VTOL PNP off the ground!
Again thank you all!
Best Regards,
RocketMoke
RocketRob

DonLakeFlyer, 2025-04-16:
Also the docs have the link the the QGC discord channel which is a great place to ask questions.

RocketMoke, 2025-04-16:
Don,
 I know Inav and Beta flight accept contributions to help cover administration costs like web hosting.....assumed it was similar with Ardupilot.

In lieu of that, as a dog lover, I am happy to help with the Painted Dog campaign!

DonLakeFlyer, 2025-04-16:
For QGC the umbrella organization which covers things like web hosting and so forth is DroneCode.org which also covers PX4 Pro firmware. Although QGC works with ArduPilot it isn't part of the ArduPilot organization.

closed 2025-04-22 by HTRamsey