mavlink/qgroundcontrol #12675: Error when building QGC on ubuntu 20.04

kronee0516, 2025-04-16:
### Your Question

Hi,

i tried to build QGC on ubuntu 20.04 
following the instruction on https://docs.qgroundcontrol.com/master/en/qgc-dev-guide/getting_started/index.html
but still got an error 

full log in the attachment
[image]

[QGC_BUILD_LOG.txt](https://github.com/user-attachments/files/19769148/QGC_BUILD_LOG.txt)

kronee0516, 2025-04-16:
solved . 

added #include <unistd.h> on the top of main.cc

DonLakeFlyer, 2025-04-16:
This should work without changing the source. Can you submit a pull with the fix you came up with?

kronee0516, 2025-04-16:
> This should work without changing the source. Can you submit a pull with the fix you came up with?

Sry, i don't quite understand how to submit a pull 
should i just submit the whole thing with built stuff so you guys can take a look why i need to add that include at the beginning of main.cc to make it work?

DonLakeFlyer, 2025-04-16:
https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/proposing-changes-to-your-work-with-pull-requests/creating-a-pull-request

PR #12697 (2025-04-19): Linux: Add unistd Include
Fix #12675
closed 2025-04-19 by HTRamsey