mavlink/qgroundcontrol #12676: Is there a function to simulate the vehicle mavlink output to uart?

kronee0516, 2025-04-16:
### Your Question

Hi,

i want to simulate the mavlink output to uart
i now  can use pymavlink to randomly send out the message 
but i think it is not so realistic
so i found this and seems mock link may do such a thing

however, when i built the qgc , i can't find how to do what i want
i set comm link to uart ,but i can't find how to use it

and i go to mock link activated a vehicle
it seems has some kind of data shown on the qgc but still i don't know how to forward to uart

and the mavlink inspector seems to have mavlink message count too

DonLakeFlyer, 2025-04-16:
What exactly are you trying to do? Why are you trying to send a mavlink message touart, for what purpose?

kronee0516, 2025-04-16:
> What exactly are you trying to do? Why are you trying to send a mavlink message touart, for what purpose?

Hi ,

Thanks for trying to help me out
i am trying to do a osd thing base on the mavlink output , like voltage, position, yaw angle ....etc
so i want to have a simulated output to debug

DonLakeFlyer, 2025-04-16:
An OSD which displays where? Separate app?

kronee0516, 2025-04-16:
like a device with monitor

DonLakeFlyer, 2025-04-16:
QGC/Application Settings/Telemetry - MAVLink Forwarding will forward to a port. You could use either SITL or MockLink to simulate a vehicle. If you need to get that mavlink to a UART you could use mavlink router (search for it) to route from port to UART.

closed 2025-04-16 by DonLakeFlyer