mavlink/qgroundcontrol #12682: Using Buttons as Joystick Input for Calibration

devjin9, 2025-04-17:
### Your Question

Hello,

I'm using a controller that only has one physical joystick.
To complete the calibration process, I would like to use buttons as substitutes for a second joystick.

Specifically, I want to:

Use the joystick for throttle and rudder calibration

Use four buttons to emulate elevator and aileron input

My question is:
Is it possible to configure button inputs to act like joystick axes during calibration?
If so, what would be the best way to implement this?

Any guidance or suggestions would be greatly appreciated.
Thank you!

-- I used ChatGPT to help me write this question.

DonLakeFlyer, 2025-04-17:
Not possible using current QGC. As far as changing the code to support this it would be a complex undertaking. You'll need to modify the joystick code to emulate stick axes from buttons.

closed 2025-04-17 by DonLakeFlyer

devjin9, 2025-04-22:
I see. Thank you for the kind response. I'll try adjusting the joystick settings.