mavlink/qgroundcontrol #12683: The latest version v4.4.4 doesn't work on my x86 MAC

kevinuav, 2025-04-17:
### Expected Behavior

I tried the latest v4.4.4 QGC on my x86 MAC(sequoia 15.3.1). It just crashes on the open stage. But v4.4.3 works well. Does QGC doesn't support x86 architecture any more?

[image]

DonLakeFlyer, 2025-04-17:
@julianoes Looks like 4.4.4 doesn't work on Mac due to a gstreamer problem? Seems like I should remove the version bump file for 4.4.4 from S3 since it's not working.

@kevinuav We are currently working towards releasing 5.0 which is based on daily builds. I believe that version should work for you.

kevinuav, 2025-04-18:
Got it . Thank you.

> [@julianoes](https://github.com/julianoes) Looks like 4.4.4 doesn't work on Mac due to a gstreamer problem? Seems like I should remove the version bump file for 4.4.4 from S3 since it's not [...]

HTRamsey, 2025-04-29:
fixed, just need to publish new release

closed 2025-04-29 by HTRamsey