mavlink/qgroundcontrol #12741: VideoReceiverLog: GStreamer error: Could not initialize window system

LuciferYJ, 2025-04-27:
### Your Question

Operating System:ubuntu20.04
QGC Version:

[image]
Qt Version:6.8.3
GStreamer Version:1.22.12.

[image]

HTRamsey, 2025-04-27:
We don't officially support Ubuntu 20.04 anymore. if you are using gstreamer 1.22 on ubuntu 20.04 then you had to have built gstreamer yourself, which means your build is missing dependencies.

closed 2025-04-27 by HTRamsey

LuciferYJ, 2025-04-27:
> We don't officially support Ubuntu 20.04 anymore. if you are using gstreamer 1.22 on ubuntu 20.04 then you had to have built gstreamer yourself, which means your build is likely missing dependencies.

In fact, I did use the source code to compile gstreamer.So which version of gstreamer do I need to use now to adapt to the qgc master branch,or do I need to upgrade to Ubuntu system?