mavlink/qgroundcontrol #12744: MacOS Crash On Startup

ehenry344, 2025-04-28:
### Description  **(required)**

Unable to launch QGC 4.4.4 using MacOS Sequoia 14.5.1 on M1 MacBook Pro

Occasionally the app hangs indefinitely and doesn't produce any logs, and other times it provides a crash log with the following termination reason.

Termination Reason:    Namespace DYLD, Code 1 Library missing
Library not loaded: /Library/Frameworks/GStreamer.framework/Versions/1.0/lib/libz.1.dylib

### System Information

QGC Version: 4.4.4
OS: MacOSX Sequoia 14.5.1
System: M1 MacBook Pro

HTRamsey, 2025-04-28:
@DonLakeFlyer @julianoes here's the reason for the 4.4.4 crash on mac

julianoes, 2025-04-29:
Is this what got fixed by @farhangnaderi in https://github.com/mavlink/qgroundcontrol/pull/12614 and now needs a release?

farhangnaderi, 2025-04-29:
> Is this what got fixed by @farhangnaderi in https://github.com/mavlink/qgroundcontrol/pull/12614 and now needs a release?

Yes. We fixed this. I and @mrpollo have tested it as well. 

cc: @DonLakeFlyer

julianoes, 2025-04-29:
Ok, so it sounds like I should trigger a release.