mavlink/qgroundcontrol #12749: Segfault when opening 3D view with specific OSM files

mrschick, 2025-04-28:
### Description  **(required)**

**Expected Behavior**
When opening the 3D view, an invalid/missing OSM file should result in an empty view, with a corresponding error message/log explaining what is wrong.

**Current Behavior**
Certain OSM files ([example](https://www.openstreetmap.org/export#map=15/46.47602/11.31862)) cause an immediate CTD due to a segmentation fault when opening the 3D view.

Debugging indicates that it occurs inside this function
https://github.com/mavlink/qgroundcontrol/blob/3e16874a20f052e7d5a4d2c31daadbd3d5601fd3/src/Viewer3D/OsmParserThread.cc#L292

### System Information

- QGC build: latest master branch, self-built from source
- Operating System: Ubuntu 24.04.2 LTS
- Flight Controller: [PX4 4001 simulated locally](https://docs.px4.io/main/en/sim_gazebo_gz/#running-the-simulation)
- Autopilot Software: PX4 master branch

### Log Files and Screenshots

- QGC Console Log:
```
Loading the OSM file!!!
3D viewer gps reference set by vehicles: 46.4789 11.3321 0
run.sh: line 2: 54210 Segmentation fault      (core dumped) ./build/Debug/QGroundControl
```

HTRamsey, 2025-04-28:
@omid-esrafilian

omid-esrafilian, 2025-04-28:
@mrschick 

Thanks for reporting the issue. 
To help me recreate the problem you encountered, could you please share your OSM file with me? The OpenStreetMap website does not allow me to download the example you mentioned.

Thanks

mrschick, 2025-04-28:
I've noticed that sometimes the export button stops working for some reason, usually refreshing the page fixes it.
Either way, here's the file [3dviewer-issue-12749.osm.txt](https://github.com/user-attachments/files/19949026/3dviewer-issue-12749.osm.txt).

omid-esrafilian, 2025-04-29:
@mrschick Thanks for sharing the OSM file.
I figured out the issue, and it should be fixed in the new pull request.

closed 2025-04-29 by HTRamsey

mrschick, 2025-04-30:
Yup, fixed after the merge. Many thanks 👍

PR #12753 (2025-04-29): Bug fix for the 3D viewer OSM parser
Fixing a bug regarding parsing OSM files for the 3D viewer.
Closes #12749