mavlink/qgroundcontrol #12761: QGroundControl v4.4.4 breaks when booting on MacOS

covelus, 2025-05-01:
### Description

**Expected Behavior**
Previous version booted normally, this one should also.

**Current Behavior**
This vesion fails with a "QGroundControl quit unexpectedly"

**Steps To Reproduce**
1. MacOS 12.7.6 Intel
2. Install QGroundControl v.4.4.4 overwriting the previously installed v.4.4.3.
3. Launch it with or without any flight controller connected.

**Additional Details**
- Report text:
`
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               QGroundControl [3108]
Path:                  /Applications/QGroundControl.app/Contents/MacOS/QGroundControl
Identifier:            org.qgroundcontrol.QGroundControl
Version:               4.4.4 (4.4.4)
Code Type:             X86-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2025-05-01 14:21:27.0199 +0200
OS Version:            macOS 12.7.6 (21H1320)
Report Version:        12
Anonymous UUID:        AE40912D-09FE-F983-318C-5CE00ECD8FC8

Sleep/Wake UUID:       E377CDDF-EAF6-497F-9C02-3E4177434E28

Time Awake Since Boot: 17000 seconds
Time Since Wake:       688 seconds

System Integrity Protection: disabled

Crashed Thread:        0

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    Namespace DYLD, Code 1 Library missing
Library not loaded: '/Library/Frameworks/GStreamer.framework/Versions/1.0/lib/libgstbase-1.0.0.dylib'
Referenced from: '/Applications/QGroundControl.app/Contents/MacOS/QGroundControl'
Reason: tried: '/Library/Frameworks/GStreamer.framework/Versions/1.0/lib/libgstbase-1.0.0.dylib' (no such file), '' (no such file), '' (no such file)
(terminated at launch; ignore backtrace)

Thread 0 Crashed:
0   dyld                          	       0x10638d9a2 __abort_with_payload + 10
1   dyld                          	       0x1063a3ab2 abort_with_payload_wrapper_internal + 80
2   dyld                          	       0x1063a3ae4 abort_with_payload + 9
3   dyld                          	       0x106351699 dyld4::halt(char const*) + 615
4   dyld                          	       0x10634d5b5 dyld4::prepare(dyld4::APIs&, dyld3::MachOAnalyzer const*) + 3995
5   dyld                          	       0x10634c4e4 start + 388

Thread 0 crashed with X86 Thread State (64-bit):
  rax: 0x0000000002000209  rbx: 0x0000000000000000  rcx: 0x00007ff7bddcc538  rdx: 0x00007ff7bddcc9a0
  rdi: 0x0000000000000006  rsi: 0x0000000000000001  rbp: 0x00007ff7bddcc580  rsp: 0x00007ff7bddcc538
   r8: 0x00007ff7bddcc5a0   r9: 0x0000000000000000  r10: 0x00000000000000a3  r11: 0x0000000000000246
  r12: 0x00000000000000a3  r13: 0x00007ff7bddcc9a0  r14: 0x0000000000000001  r15: 0x0000000000000006
  rip: 0x000000010638d9a2  rfl: 0x0000000000000246  cr2: 0x0000000000000000
  
Logical CPU:     0
Error Code:      0x02000209 
Trap Number:     133
[... 122 lines cut ...]
}

Model: MacBookPro12,1, BootROM 489.0.0.0.0, 2 processors, Dual-Core Intel Core i5, 2,9 GHz, 16 GB, SMC 2.28f7
Graphics: Intel Iris Graphics 6100, Intel Iris Graphics 6100, Built-In
Display: Color LCD, 2560 x 1600 Retina, Main, MirrorOff, Online
Display: VW222, 1680 x 1050 (Widescreen Super eXtended Graphics Array Plus), MirrorOff, Online
Memory Module: BANK 0/DIMM0, 8 GB, DDR3, 1867 MHz, 0x02FE, 0x4544464232333241314D412D4A442D460000
Memory Module: BANK 1/DIMM0, 8 GB, DDR3, 1867 MHz, 0x02FE, 0x4544464232333241314D412D4A442D460000
AirPort: spairport_wireless_card_type_wifi (0x14E4, 0x133), Broadcom BCM43xx 1.0 (7.77.111.1 AirPortDriverBrcmNIC-1710.4)
AirPort: 
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
USB Device: USB30Bus
USB Device: Bluetooth USB Host Controller
USB Device: PX4 FMU v6X.x
Thunderbolt Bus: MacBook Pro, Apple Inc., 27.1
`

### System Information

- QGC Version: [e.g. 4.4.0] **(required)**
- QGC build: [e.g. daily, stable, self-built from source, etc...]
- Operating System: [e.g. Windows 11, Ubuntu 22.04, macOS 15, iOS 17 ]
- Flight Controller: [e.g. CubePilot Cube Orange, Pixhawk 6X, etc.]
- Autopilot Software (with version): [e.g., PX4 1.15.4, ArduCopter 4.5.7 ]

HTRamsey, 2025-05-01:
Yup I believe it's been fixed but a new release needs to be published

HTRamsey, 2025-05-18:
o with v5.0 now having an rc, v4.4.4 macos won't likely be fixed

closed 2025-05-18 by HTRamsey