mavlink/qgroundcontrol #12765: Compass Calibration is not working

Gowthamk43, 2025-05-02:
### Description

**Expected Behavior**
when i click to calibrate it should calibrate the sensors like compass
**Current Behavior**
when i start calibrating it shows downloading but compass not calibrated it shows red in v4.4.4 is working good in daily update i get error in Compass

**Steps To Reproduce**
1.
2.
3.

**Additional Details**
Add any extra info (errors, warnings, etc.) here.

### System Information

- QGC Version: [Daily Version] **(required)**
- QGC build: [daily]
- Operating System:[Windows 11, Ubuntu 22.04]
- Flight Controller: [ CubePilot Cube Orange.]
- Autopilot Software (with version): [e.g., PX4 1.15.4, ArduCopter 4.5.7 ]

HTRamsey, 2025-05-02:
I'll check this weekend

HTRamsey, 2025-05-02:
is this for ardupilot?

Gowthamk43, 2025-05-03:
yes

closed 2025-05-03 by HTRamsey

Gowthamk43, 2025-05-04:
Error is resolved ?

PR #12767 (2025-05-03): Fix APM Compass Callibration
closes #12765