mavlink/qgroundcontrol #12773: Joystick doesn't work (xbox series x)

dakejahl, 2025-05-04:
### Description

Controller doesn't work, sticks don't move in the Joystick menu.

### System Information

- QGC main as of 5/4/2025
- PX4 main with gz sim -- `make px4_sitl gz_x500`
- XBox Series X controller connected via USB C

It works with QGC 4.4

HTRamsey, 2025-05-05:
Three options I can think of:

- Add a custom mapping [here](https://github.com/mavlink/qgroundcontrol/blob/397d78109c5bb2e00a6a968cff58e774be67d3b4/src/Joystick/CMakeLists.txt#L97)
- Update to SDL3
- Adjust build options [here](https://github.com/mavlink/qgroundcontrol/blob/397d78109c5bb2e00a6a968cff58e774be67d3b4/src/Joystick/CMakeLists.txt#L59)

dakejahl, 2025-05-05:
This checkbox was not selected 🤦 

[image]

closed 2025-05-05 by dakejahl