mavlink/qgroundcontrol #12788: Unable to set Joystrick Axis Frequency

er-jk, 2025-05-06:
### Description

**Expected Behavior**
Whenever the axis frequency changes by user from UI vehicle settings -> Joystick -> Advanced, it needs to change based on min & max frequency of 0.25f to 200.0f. 

**Current Behavior**
Current it's not changing the value because the minimum and maximum frequency value seems to be same. Initial maximum value is keeps as _minAxisFrequencyHz

### System Information

- QGC Version: [e.g. 4.4.0]
- Operating System: Windows 11
- Flight Controller: ArduCopter
- Autopilot Software (with version): ArduCopter 4.5.7 

### Log Files and Screenshots

After changing the Q_PROPERTY of maxAxisFrequencyHz property, it's working fine for me!

**Current Code**
`Q_PROPERTY(float    maxAxisFrequencyHz      MEMBER _minAxisFrequencyHz                              CONSTANT)`

**After Change of code (It is Working)**
`Q_PROPERTY(float    maxAxisFrequencyHz      MEMBER _maxAxisFrequencyHz                              CONSTANT)`

Please review this change and share any feedback you may have.
Thank you for your support — you’re doing an excellent job maintaining this project.
Looking forward to your response!

HTRamsey, 2025-05-06:
I think this is fixed in the master build

er-jk, 2025-05-06:
> I think this is fixed in the master build

Cool, I was using Stable_V4.4!

HTRamsey, 2025-05-07:
Since v5 is about to come out and contains this fix, I doubt v4.4 will get updated with this fix so I'm going to mark this completed

closed 2025-05-07 by HTRamsey