mavlink/qgroundcontrol #12800: .exe file does not seem to execute after windeployqt

vorsprungdurchtae, 2025-05-08:
### Your Question

Hi all,

does anyone already experience that qgroundcontrol.exe does not appear to execute after windeployqt without error?

cromwellov, 2025-05-08:
I am having same issue, especially when I build for Release

HTRamsey, 2025-05-08:
Do you mean it doesn't execute or it has an error

cromwellov, 2025-05-08:
> Do you mean it doesn't execute or it has an error

It tries to execute but leaves this error: 16:31:17: The process crashed.

HTRamsey, 2025-05-08:
Are you building the same way the CI workflow does? Does the release build artifact downloaded from GitHub work?

vorsprungdurchtae, 2025-05-09:
In my case, the .exe file generated by Release function in Qt Creator does not execute. Even though I also executed the windeployqt.

HTRamsey, 2025-05-09:
You aren't running windeployqt manually right?

vorsprungdurchtae, 2025-05-09:
I ran it by the followings in cmd:

path\to\qgc.exe> windeployqt qgroundcontrol.exe

HTRamsey, 2025-05-09:
Oh thats probably why you're having problems then. Make sure to follow how the CI workflow builds

vorsprungdurchtae, 2025-05-09:
Ok, can you please tell me where I can find Cl workflow build guid?

HTRamsey, 2025-05-09:
This is how our auto builds are done: https://github.com/mavlink/qgroundcontrol/blob/master/.github%2Fworkflows%2Fwindows.yml

closed 2025-05-13 by HTRamsey