mavlink/qgroundcontrol #12803: Error during execution of cmake -DCMAKE_BUILD_TYPE=Release -G Ninja ..

vorsprungdurchtae, 2025-05-09:
### Your Question

Hi,

I meanwhile am building the release of QGC on local using Ninja. And I got the following error:

-- QGroundControl version: v4.4.4
-- Found PkgConfig: C:/MinGW/bin/pkg-config.exe (found version "0.26")
-- Checking for modules 'gstreamer-1.0>=1.14;gstreamer-video-1.0>=1.14;gstreamer-gl-1.0>=1.14;egl'
--   No package 'gstreamer-1.0' found
--   No package 'gstreamer-video-1.0' found
--   No package 'gstreamer-gl-1.0' found
--   No package 'egl' found
CMake Error at CMakeLists.txt:123 (find_package):
  Could not find a package configuration file provided by "Qt5" with any of
  the following names:

    Qt5Config.cmake
    qt5-config.cmake

I got this error regarding gstreamer even though I did not get such error during build on Qt Creator.
And I want to know how to solve the error regarding Qt5Config.cmake and qt5-config.cmake.

I am currently using Qt 6.8.2 MSVC 2022

HTRamsey, 2025-05-09:
QGC v4.4 uses Qt5.15 not Qt6

vorsprungdurchtae, 2025-05-11:
Then is Qt5.15.x is more proper than Qt6.x? also for debug and release?

HTRamsey, 2025-05-11:
Yes, it's not optional. You have to use the same Qt version as the one QGC was designed with.

vorsprungdurchtae, 2025-05-12:
I am now again trying to build using Qt 5.15.2.

Now I got the following errors:

1. Project ERROR: You need to set the ANDROID_NDK_ROOT environment variable to point to your Android NDK.
Could not read qmake configuration file C:/Qt/Qt5.15.2/5.15.2/android/mkspecs/android-clang/qmake.conf.

and 

2. CMake Error ... is not able to compile a simple test program.
...
CMAKE_MT-NOTFOUND
...
rc /fo ... failed (exit code 0) with the following output:
no such file or directory.

I hope someone advise me how to address this issue.

The whole error message is : 

CMake Error at C:/Program Files/CMake/share/cmake-4.0/Modules/CMakeTestCCompiler.cmake:67 (message):
  The C compiler

    "C:/Program Files (x86)/Microsoft Visual Studio 14.0/VC/BIN/x86_amd64/cl.exe"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: 'C:/Users/ttss0/AppData/Local/Temp/QtCreator-oZqkkj/qtc-cmake-vVCqCRxA/CMakeFiles/CMakeScratch/TryCompile-6r01y4'
    
    Run Build Command(s): C:/ninja-win/ninja.exe -v cmTC_1272b
    [1/2] C:\PROGRA~2\MICROS~1.0\VC\BIN\X86_AM~1\cl.exe  /nologo   /DWIN32 /D_WINDOWS  /Ob0 /Od /RTC1 -MDd -Zi /showIncludes /FoCMakeFiles\cmTC_1272b.dir\testCCompiler.c.obj /FdCMakeFiles\cmTC_1272b.dir\ /FS -c C:\Users\ttss0\AppData\Local\Temp\QtCreator-oZqkkj\qtc-cmake-vVCqCRxA\CMakeFiles\CMakeScratch\TryCompile-6r01y4\testCCompiler.c
    [2/2] C:\Windows\system32\cmd.exe /C "cd . && "C:\Program Files\CMake\bin\cmake.exe" -E vs_link_exe --msvc-ver=1900 --intdir=CMakeFiles\cmTC_1272b.dir --rc=rc --mt=CMAKE_MT-NOTFOUND --manifests  -- C:\PROGRA~2\MICROS~1.0\VC\bin\X86_AM~1\link.exe /nologo CMakeFiles\cmTC_1272b.dir\testCCompiler.c.obj  /out:cmTC_1272b.exe /implib:cmTC_1272b.lib /pdb:cmTC_1272b.pdb /version:0.0 /machine:x64  /debug /INCREMENTAL /subsystem:console  kernel32.lib user32.lib gdi32.lib winspool.lib shell32.lib ole32.lib oleaut32.lib uuid.lib comdlg32.lib advapi32.lib && cd ."
    FAILED: cmTC_1272b.exe 
    C:\Windows\system32\cmd.exe /C "cd . && "C:\Program Files\CMake\bin\cmake.exe" -E vs_link_exe --msvc-ver=1900 --intdir=CMakeFiles\cmTC_1272b.dir --rc=rc --mt=CMAKE_MT-NOTFOUND --manifests  -- C:\PROGRA~2\MICROS~1.0\VC\bin\X86_AM~1\link.exe /nologo CMakeFiles\cmTC_1272b.dir\testCCompiler.c.obj  /out:cmTC_1272b.exe /implib:cmTC_1272b.lib /pdb:cmTC_1272b.pdb /version:0.0 /machine:x64  /debug /INCREMENTAL /subsystem:console  kernel32.lib user32.lib gdi32.lib winspool.lib shell32.lib ole32.lib oleaut32.lib uuid.lib comdlg32.lib advapi32.lib && cd ."
    RC Pass 1: command "rc /fo CMakeFiles\cmTC_1272b.dir/manifest.res CMakeFiles\cmTC_1272b.dir/manifest.rc" failed (exit code 0) with the following output:
    no such file or directory
    ninja: build stopped: subcommand failed.
    
    

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:44 (project)

-- Configuring incomplete, errors occurred!
CMake process exited with exit code 1.
Elapsed time: 00:02.
Project ERROR: You need to set the ANDROID_NDK_ROOT environment variable to point to your Android NDK.
Could not read qmake configuration file C:/Qt/Qt5.15.2/5.15.2/android/mkspecs/android-clang/qmake.conf.

tranhien1612, 2025-05-12:
Hi, I am using Qt 5.15 with `gstreamer-1.0-android-universal-1.18.6` to build for android. I added element gstreamer "flvmux" and "rtmp" to sourcecode. but i had problem. I don't know if this error is caused by the gstreamer version or not. pls help me fix it.
```
gst_element_factory_make('flvmux') failed
gst_element_factory_make('rtmpsink') failed
```

tranhien1612, 2025-05-13:
@vorsprungdurchtae with QT 5.15, pls try with qgc branch v4.3.

vorsprungdurchtae, 2025-05-13:
> Yes, it's not optional. You have to use the same Qt version as the one QGC was designed with.

However, I want to know why all the cmakelists.txt of QGC4.4.4 are designed to run on qt6. There appear a bunch of Qt6 specific commands, such as Qt6::Core which I had to replace to Qt5::Core

HTRamsey, 2025-05-13:
The master branch is on Qt6.8, the latest stable (v4.4.x) is on Qt5.15. For v4.4 it is better to be using qmake instead of cmake

closed 2025-05-13 by HTRamsey