mavlink/qgroundcontrol #12806: QGC v5.0.0 pre-release testing coordination

mrpollo, 2025-05-09:
## 🚀 Help Us Finalize the QGroundControl v5.0 Stable Release

Hey folks! We're getting close to launching the **v5.0 stable release** — in the next few weeks (maybe even days!) — and we need your help to squash the last remaining bugs.

We're entering a **feature freeze**, and development will transition to a new release branch focused solely on **bug fixes and stabilization**. This is your chance to help us ensure v5.0 is solid for everyone.

We'd really appreciate it if you could test the app in a real-world environment and report any issues or regressions you notice compared to previous releases.

---

### 📣 How to Report Issues

- Clearly mention the **version** you're testing and your **operating system**.
- Describe your hardware setup (telemetry, flight controller, firmware).
- Include **detailed steps to reproduce** the issue.
- Add **screenshots or screen recordings** where possible — they really help!
- Tag your issue with the label: `Release: v5.0`

---

### Feel free to drop feedback below

If you are unsure about creating an issue, feel free to drop a comment below with your feedback.

## Thanks in advance for your support and feedback.

**Safe flights! 🛫**

TeddyZaremba, 2025-05-09:
Clearly mention the version you're testing and your operating system.
QGC 424db48 
ubuntu 22.04

Describe your hardware setup (telemetry, flight controller, firmware).
joystick, PX4 1.16, simulation (gz_x500)

Include detailed steps to reproduce the issue.
Have a joystick connected without enabling it. Warning for joystick not connected says you can enable it by modify COM_RC_IN_MODE, but doesn't mention you need to go into vehicle configuration to actually enable it (somewhat misleading). 

Add screenshots or screen recordings where possible — they really help!

[image]

Tag your issue with the label: Release: v5.0

jnomikos, 2025-05-13:
https://github.com/mavlink/qgroundcontrol/issues/12806

Apologies, did not see the Release 5.0 tag and I don't see the option to change the tag after I made the issue.

DonLakeFlyer, 2025-05-14:
> Have a joystick connected without enabling it. Warning for joystick not connected says you can enable it by modify COM_RC_IN_MODE, but doesn't mention you need to go into vehicle configuration to actually enable it (somewhat misleading).

Nothing has changed here. This message is from the vehicle firmware. It's just telling you that it isn't getting control inputs from anywhere. It doesn't really mean joystick specifically. It can also mean RC transmitter sticks. You'd need to file a bug against PX4 Firmware if it's confusing.

dirksavage88, 2025-05-16:
I'm not getting the control allocation page to show up in the actuators/motors tab. It pulls up just fine in qgc 4.4.3. Metadata transfer timeout?

[image]

DonLakeFlyer, 2025-05-17:
@dirksavage88 Can you create an Issue and assign it to me with details. Can't quite remember exactly what triggers the actuator stuff now. There should be some logging in the initial connect sequence code to turn on. Need to find it.

HTRamsey, 2025-05-17:
FWIW Just did it over serial

[image]

kschoelpple, 2025-05-19:
Version: 5.0.1
System: Kubuntu 24.04

Problem: Any page requiring a scroll bar, will have scroll bar fade out after 3 seconds. ( This was already observed in previous daily build versions). 
This won't allow any changes to be made once scroll bar is faded out. - However, switching to another page and then back will briefly restore scroll bar.
Tag: Release 5.0.1

[image]

Aacis017, 2025-05-22:
I'm experiencing noticeable delay in RTSP Video Stream . 
I’m using the following setup:
1.Cube Orange+
2.SIYI MK15E remote controller(Android 9)
3.SIYI A8 Mini camera

Jan-X, 2025-05-28:
Hi,
would it be possible to also get an Android 64 Bit release version ARM64 APK version of the QGC 5.0 preview (like you have published with the GQC 4.4.5 "QGroundControl64.apk" ?
Thanks.

kschoelpple, 2025-06-08:
Hi,
Lenovo Tap M5
QGC v.5.0.2
Using the .apk file, won't start after installation.

Is it possible to get a 32bit version?

DonLakeFlyer, 2025-06-08:
@kschoelpple APK is both 64 and 32 together. Can you create a separate Issue with details?

kschoelpple, 2025-06-09:
Hi,
SIYI MK15 E
Android 9
.apk file won't install (v.5.0.2)  - previous stable versions did install

Have developer option active. Also tried standard. Tried loading from internal memory - Download folder. 
Also copied onto SD card. - Both options fail.
Also of "Allow install from unknown sources" set.

....did notice Aacis017 must have somehow been successful with installation on MK15E.

Aacis017, 2025-06-09:
> Hi, SIYI MK15 E Android 9 .apk file won't install (v.5.0.2) - previous stable versions did install
>
> Have developer option active. Also tried standard. Tried loading from internal memory - Download [...]

It failed for me too on the MK15E (Android 9) when trying to install the v5.0.2 .apk.

However, I built the project from source using Qt Creator and released the APK myself.

Aacis017, 2025-06-09:
i think 32-bit  .apk file resolves this issue

> But where do we stand here? Anyone on Android 9 has to build their own .apk from source, going forward? ....not everyone has the ability to do so.
> 
> (Thanks, appreciate the support and effort from everyone going into this new version.)

HTRamsey, 2025-06-09:
Is that device armeabi-v7a? If so, the released APK should support that

kschoelpple, 2025-06-09:
@HTRamsey 

Happy to report that version (your link provided) worked.

Thanks

Jan-X, 2025-06-09:
> Hi, would it be possible to also get an Android 64 Bit release version ARM64 APK version of the QGC 5.0 preview (like you have published with the GQC 4.4.5 "QGroundControl64.apk" ? Thanks.

Okay, got the QGC 5.0 RC 3 launching on a Pico 4 Ultra VR Glasses (as a 2D Android app).
It seems like to work, unfortunately the mapping of the VR controllers is not perfect (yet).
But it seems to work....

Muhyeng98, 2025-06-16:
> I'm experiencing noticeable delay in RTSP Video Stream . I’m using the following setup: 1.Cube Orange+ 2.SIYI MK15E remote controller(Android 9) 3.SIYI A8 Mini camera

Have you solved the issue? I'm also experiencing the same problem. It's beyond a simple delay and has become unusable.

DonLakeFlyer, 2025-06-18:
> QGC5 RC3 terminates on Linux ...

Seems to be working fine on Ubuntu 24.04. You might need to build yourself and get this in a debugger for us. We're gonna have a hard time reproing.

HTRamsey, 2025-06-18:
Just merged a crash fix for mission stuff, maybe that was it?

kschoelpple, 2025-06-21:
Hi,
regarding previously mentioned alignment error of Compass on Kubuntu:
Unfortunately worse in latest version.
v.5.0.3
System: Kubuntu 24.04
KDE Plasma: 5.27.12
Graphics Platform: X11

[image]

HTRamsey, 2025-06-21:
Somebody else apparently had good results when using GStreamer 1.26 for android [here](#12995), it really seems to me the lag is limited to only certain cameras/streams on Android which makes it really hard for me to reproduce and solve

DonLakeFlyer, 2025-06-21:
@kschoelpple Just fixed yet another bug with respect to HUD positioning. Hopefully, last one!

kschoelpple, 2025-06-27:
Hi,
Android 15
QGC v.5.0.4

HUD side by side not displaying correctly.

[image]

DonLakeFlyer, 2025-06-27:
> HUD side by side not displaying correctly.

@kschoelpple Not reproing for me. Take a look at Console logs to see if anything strange it showing up in there with some sort of qml error.

kschoelpple, 2025-06-27:
No qml error, but a whole heap in regards to display and missing library.

DonLakeFlyer, 2025-06-28:
@kschoelpple I'm talking about the output from the QGC Console page.

kschoelpple, 2025-06-29:
0.294 - debug: Settings location "/data/user/0/org.mavlink.qgroundcontrol/files/settings/QGroundControl/QGroundControl.ini" Is writable?: true (qgc.qgcapplication:unknown:0)
     0.310 - debug: Filter rules "*Log.debug=false\nqgc.*.debug=false\nqgc.android.androidinit.debug=true\nqgc.android.androidserial.debug=true\nqgc.api.qgccoreplugin.debug=true\nqgc.api.qgcflyviewoptions.debug=true\nqgc.api.qgcoptions.debug=true\nqgc.api.qmlcomponentinfo.debug=true\nEventsLog.debug=true\nComponentInformationCacheLog.debug=true\nqgc.videomanager.videoreceiver.gstreamer.api.debug=true\nqt.qml.connections=false" (qgc.utilities.qgcloggingcategory:unknown:0)
     3.154 - warning: availableDevicesInfo returned null (qgc.android.androidserial:unknown:0)
     4.098 - warning: availableDevicesInfo returned null (qgc.android.androidserial:unknown:0)
     5.048 - warning: availableDevicesInfo returned null (qgc.android.androidserial:unknown:0)
     5.998 - warning: availableDevicesInfo returned null (qgc.android.androidserial:unknown:0)
     6.988 - warning: availableDevicesInfo returned null (qgc.android.androidserial:unknown:0)
     7.990 - warning: availableDevicesInfo returned null (qgc.android.androidserial:unknown:0)
     8.989 - warning: availableDevicesInfo returned null (qgc.android.androidserial:unknown:0)
     9.989 - warning: availableDevicesInfo returned null (qgc.android.androidserial:unknown:0)
    10.989 - warning: availableDevicesInfo returned null (qgc.android.androidserial:unknown:0)
    11.989 - warning: availableDevicesInfo returned null (qgc.android.androidserial:unknown:0)
    12.988 - warning: availableDevicesInfo returned null (qgc.android.androidserial:unknown:0)
    13.990 - warning: availableDevicesInfo returned null (qgc.android.androidserial:unknown:0)
    14.990 - warning: availableDevicesInfo returned null (qgc.android.androidserial:unknown:0)
    15.990 - warning: availableDevicesInfo returned null (qgc.android.androidserial:unknown:0)
    16.990 - warning: availableDevicesInfo returned null (qgc.android.androidserial:unknown:0)
    16.997 - warning: qrc:/qml/QGroundControl/Controls/AppMessages.qml:47:25: Unable to assign [undefined] to QString (default:unknown:47)
    17.010 - warning: qrc:/qml/QGroundControl/Controls/AppMessages.qml:47:25: Unable to assign [undefined] to QString (default:unknown:47)
    17.024 - warning: qrc:/qml/QGroundControl/Controls/AppMessages.qml:47:25: Unable to assign [undefined] to QString (default:unknown:47)
    17.033 - warning: qrc:/qml/QGroundControl/Controls/AppMessages.qml:47:25: Unable to assign [undefined] to QString (default:unknown:47)
    17.042 - warning: qrc:/qml/QGroundControl/Controls/AppMessages.qml:47:25: Unable to assign [undefined] to QString (default:unknown:47)
    17.052 - warning: qrc:/qml/QGroundControl/Controls/AppMessages.qml:47:25: Unable to assign [undefined] to QString (default:unknown:47)
    17.060 - warning: qrc:/qml/QGroundControl/Controls/AppMessages.qml:47:25: Unable to assign [undefined] to QString (default:unknown:47)
    17.069 - warning: qrc:/qml/QGroundControl/Controls/AppMessages.qml:47:25: Unable to assign [undefined] to QString (default:unknown:47)
    17.078 - warning: qrc:/qml/QGroundControl/Controls/AppMessages.qml:47:25: Unable to assign [undefined] to QString (default:unknown:47)
    17.989 - warning: availableDevicesInfo returned null (qgc.android.androidserial:unknown:0)
    18.990 - warning: availableDevicesInfo returned null (qgc.android.androidserial:unknown:0)
    19.989 - warning: availableDevicesInfo returned null (qgc.android.androidserial:unknown:0)
    20.990 - warning: availableDevicesInfo returned null (qgc.android.androidserial:unknown:0)
    21.991 - warning: availableDevicesInfo returned null (qgc.android.androidserial:unknown:0)

HTRamsey, 2025-07-01:
You can ignore all the androidserial stuff (that shouldn't have been a warning). But I looked at the AppMessages thing and it looks like it is referencing an unknown variable, but v4.4 was the same way

kschoelpple, 2025-07-01:
@HTRamsey  Thanks for the reply.
You can't push the install through on Android 15 as once you press OK - seen on that screenshot - another message pops up "Not installed".
And that's it. 
Repeating it ends up the same way.
I suspect it's linked to an Android system safety update in back in May. - Hence the v4.4 did install as was done prior to that update.

DonLakeFlyer, 2025-07-02:
> Is that supposed to respond any time

Yes. It's based on percent remaining from the battery_status mavlink message. In your picture you can see that percent remaining is still being reported as 100%. Some sort of config error. Or I vaguely remember maybe ArduPilot doesn't support percent remaining, don't know.

kschoelpple, 2025-07-11:
@DonLakeFlyer 
The cache on the Android device linked to every app.
Long press app icon --> Click on "App-Info", Select "Storage&cache" then click on "Clear cache"
That deletes any old settings linked to that app and forces a fresh configuration.
In rare cases the "Delete" storage is also required when stubborn issues persist.

kschoelpple, 2025-07-11:
Android 9 on SIYI MK15E
QGC v.5.0.5
About 3/4 into installation, installation aborts.
Message "App not installed" appears.

DonLakeFlyer, 2025-08-21:
I'm closing this. Any remaining problems should be created in separate issues.