mavlink/qgroundcontrol #12807: QGC no voice

Tfly6, 2025-05-10:
### Description

**Expected Behavior**
When launching PX4 SITL, I open QGC and can hear voice in QGC.

**Current Behavior**

I can't hear voice in QGC. I can confirm my Ubuntu has voice and can listen to music.

**Steps To Reproduce**
1.Download QGC V4.4.4
2.Launch PX4 SITL
3.Open QGC 

**Additional Details**
```bash
Settings location "/home/tfly/.config/QGroundControl.org/QGroundControl.ini" Is writable?: true
Filter rules "*Log.debug=false\nGStreamerAPILog.debug=true\nqt.qml.connections=false"
System reported locale: QLocale(English, Latin, United States) ; Name "en_US" ; Preffered (used in maps):  "en-US"
VideoReceiverLog: Stop called on empty URI
VideoReceiverLog: Stop called on empty URI
MAVLinkLogManagerLog: MAVLink logs directory: "/home/tfly/Documents/QGroundControl/Logs"
Map Cache in: "/home/tfly/.cache/QGCMapCache300" / "qgcMapCache.db"
setCurrentPlanViewSeqNum
setCurrentPlanViewSeqNum
"v4.4.4"
Adding target QHostAddress("127.0.0.1") 18570
setCurrentPlanViewSeqNum
setCurrentPlanViewSeqNum
FirmwarePluginLog: Unable to parse version info from file "https://api.github.com/repos/PX4/Firmware/releases"
ComponentInformationTranslationLog: Locale  "en_US"  not found in json
ComponentInformationTranslationLog: Locale  "en_US"  not found in json
ParameterManagerLog: Attemping load from cache
ParameterManagerLog: Parameters loaded from cache /home/tfly/.config/QGroundControl.org/ParamCache/1_1.v2
QCoreApplication::postEvent: Unexpected null receiver
setCurrentPlanViewSeqNum
setCurrentPlanViewSeqNum
VideoReceiverLog: Stop called on empty URI
VideoReceiverLog: Stop called on empty URI
Quit event true
Done
VideoReceiverLog: Stop called on empty URI
VideoReceiverLog: Stop called on empty URI
```

### System Information

- QGC Version: [4.4.0] **(required)**
- QGC build: [stable]
- Operating System: [Ubuntu 20.04]
- Flight Controller: [PX4 SITL]
- Autopilot Software (with version): [PX4 1.14.4 ]

DonLakeFlyer, 2025-05-24:
Try the latest 5.0 release candidate and see if it's fixed: https://github.com/mavlink/qgroundcontrol/releases/tag/v5.0.1

DonLakeFlyer, 2025-05-24:
QGC 5.0 only works on 22.04 and above. You'll have other upgrade if you want to use 5.0

Tfly6, 2025-05-25:
> QGC 5.0 only works on 22.04 and above. You'll have other upgrade if you want to use 5.0

Perhaps it is an issue with my Ubuntu system since QGC works fine in other operating systems.

DonLakeFlyer, 2025-08-21:
Hard to say. But at this point only 5.0 is the supported release. So we won't be fixing anything in older releases.