mavlink/qgroundcontrol #12816: Misleading Joystick Enable Warning

TeddyZaremba, 2025-05-12:
### Description

**Expected Behavior**
Describe what you thought should happen here.
Able to enable the joystick from the warning popup in the 'overall status' section

**Current Behavior**
Have a joystick connected without enabling it. Warning for joystick not connected says you can enable it by modify COM_RC_IN_MODE, but it doesn't mention you need to go into vehicle configuration to actually enable it (somewhat misleading).

**Steps To Reproduce**
1. Connect a joystick without hitting enable joystick under 'vehicle configuration/ joystick'
2. Wait for error in the 'overall status box'

**Additional Details**

### System Information

- QGC Version: QGC https://github.com/mavlink/qgroundcontrol/commit/424db4863ac24f73228a4e5b3c4e109fec87b9e7
- QGC build: daily
- Operating System: Ubuntu 22.04
- Flight Controller: gz_x500
- Autopilot Software (with version): PX4 1.16

### Log Files and Screenshots

[image]

HTRamsey, 2025-05-12:
I believe that's just a message straight from PX4 and not actually related to QGC itself, but I suppose detecting a joystick connect/disconnect event and displaying a notification about enabling it is reasonable

DonLakeFlyer, 2025-05-16:
QGC has no idea what those mavlink events mean. They are just text messages. So no way to do something intelligent. Only thing is to change message from firmware side to make it clearer. When firmware message says "sticks" it's talking about any sticks: joystick, RC transmitter, ... whatever. It's not specifically saying joystick is not enabled. Enter firmware side issue if you want to clear this up a bit.

closed 2025-05-16 by DonLakeFlyer