mavlink/qgroundcontrol #12822: QML: Auto Adjust Vehicle Component Summary Height

HTRamsey, 2025-05-13:
### Description

Height is currently set to a fixed value, but needs to autoadjust based on model height

### System Information

- QGC Version: [e.g. 4.4.0] **(required)**
- QGC build: [e.g. daily, stable, self-built from source, etc...]
- Operating System: [e.g. Windows 11, Ubuntu 22.04, macOS 15, iOS 17 ]
- Flight Controller: [e.g. CubePilot Cube Orange, Pixhawk 6X, etc.]
- Autopilot Software (with version): [e.g., PX4 1.15.4, ArduCopter 4.5.7 ]

### Log Files and Screenshots

[image]

igs-raiderler, 2025-05-13:
Great implementation of this feature. I particularly like the approach to error handling.

DonLakeFlyer, 2026-03-12:
Layout is better now, but some summary implementations are crap. Need to fix those. Examples:

[image]