mavlink/qgroundcontrol #12825: How to add and compiler gstreamer element?

tranhien1612, 2025-05-13:
### Your Question

Hi, i want to add gstreamer element more. I use `flv` and `rtmpsink` element. 

At first when I called the `gst_element_factory_make("flvmux", nullptr))` and `gst_element_factory_make("rtmpsink", nullptr))` function I got the following error message:
```
  start gst_element_factory_make('flvmux') failed
  start gst_element_factory_make('rtmpsink') failed
```

After, I added some code in file `GStreamer.cc`:
```
#if defined(__android__) || defined(__ios__)
    GST_PLUGIN_STATIC_DECLARE(coreelements);
    GST_PLUGIN_STATIC_DECLARE(playback);
    GST_PLUGIN_STATIC_DECLARE(libav);
    GST_PLUGIN_STATIC_DECLARE(rtp);
    GST_PLUGIN_STATIC_DECLARE(flv); 
    GST_PLUGIN_STATIC_DECLARE(rtmp); 
    GST_PLUGIN_STATIC_DECLARE(rtsp);
    GST_PLUGIN_STATIC_DECLARE(udp);
    GST_PLUGIN_STATIC_DECLARE(videoparsersbad);
    GST_PLUGIN_STATIC_DECLARE(x264);
    GST_PLUGIN_STATIC_DECLARE(rtpmanager);
    GST_PLUGIN_STATIC_DECLARE(isomp4);
    GST_PLUGIN_STATIC_DECLARE(matroska);
    GST_PLUGIN_STATIC_DECLARE(mpegtsdemux);
    GST_PLUGIN_STATIC_DECLARE(opengl);
    GST_PLUGIN_STATIC_DECLARE(tcp);
#if defined(__android__)
    GST_PLUGIN_STATIC_DECLARE(androidmedia);
#elif defined(__ios__)
    GST_PLUGIN_STATIC_DECLARE(applemedia);
#endif
....

#if defined(__android__) || defined(__ios__)
    GST_PLUGIN_STATIC_REGISTER(coreelements);
    GST_PLUGIN_STATIC_REGISTER(playback);
    GST_PLUGIN_STATIC_REGISTER(libav);
    GST_PLUGIN_STATIC_REGISTER(rtp);
    GST_PLUGIN_STATIC_REGISTER(flv);
    GST_PLUGIN_STATIC_REGISTER(rtmp); 
    GST_PLUGIN_STATIC_REGISTER(rtsp);
    GST_PLUGIN_STATIC_REGISTER(udp);
    GST_PLUGIN_STATIC_REGISTER(videoparsersbad);
    GST_PLUGIN_STATIC_REGISTER(x264);
    GST_PLUGIN_STATIC_REGISTER(rtpmanager);
    GST_PLUGIN_STATIC_REGISTER(isomp4);
    GST_PLUGIN_STATIC_REGISTER(matroska);
    GST_PLUGIN_STATIC_REGISTER(mpegtsdemux);
    GST_PLUGIN_STATIC_REGISTER(opengl);
    GST_PLUGIN_STATIC_REGISTER(tcp);

#if defined(__android__)
    GST_PLUGIN_STATIC_REGISTER(androidmedia);
#elif defined(__ios__)
    GST_PLUGIN_STATIC_REGISTER(applemedia);
#endif
#endif
```
it seems like the "flv" element was added successfully (because i didn't error ` start gst_element_factory_make('flvmux') failed` ) . But I get an error with the "rtmpsink" element:

[image]

[image]

I don't have much knowledge about this problem, please help me deal with it.

HTRamsey, 2025-05-13:
https://github.com/mavlink/qgroundcontrol/blob/4d832cc3a6378f284c63d1e4a41a2634d382eedd/src/VideoReceiver/VideoReceiver.pri#L87

tranhien1612, 2025-05-13:
> [qgroundcontrol/src/VideoReceiver/VideoReceiver.pri](https://github.com/mavlink/qgroundcontrol/blob/4d832cc3a6378f284c63d1e4a41a2634d382eedd/src/VideoReceiver/VideoReceiver.pri#L87)
> 
> Line 87 in [4d832cc](/mavlink/qgroundcontrol/commit/4d832cc3a6378f284c63d1e4a41a2634d382eedd)
> 
>  LIBS += -L$$GST_ROOT/lib/gstreamer-1.0 \

i checked and it has that command line. at first i used gstreamer version 1.18.6 but it was buggy, so now i am switching to using version 1.20.7 for android

HTRamsey, 2025-05-13:
you added your plugins to that VideoReceiver.pri file?

tranhien1612, 2025-05-13:
@HTRamsey I have added some plugins, but I don't know if there are any missing plugins.

[image]

HTRamsey, 2025-05-13:
so you would add -lrtmp in this section https://github.com/mavlink/qgroundcontrol/blob/4d832cc3a6378f284c63d1e4a41a2634d382eedd/src/VideoReceiver/VideoReceiver.pri#L107

tranhien1612, 2025-05-13:
Thanks for your help. I will try it tomorrow since it is dark now. By the way, I want to ask the following question:
I want to create a gstreamer pipe for rtmp as follows, I tested it using command line. I am trying to add it to qgrouncontrol following the structure you are doing. Will this affect the system?
```
//                          +-->queue-->_decoderValve[-->_decoder-->_videoSink]
//                           |
// _source-->_tee +
//                           |
//                          +-->queue-->_recorderValve[-->_fileSink]
//                           |
//                           |
//                          +-->queue-->rtph264depay-->h264parse-->flvmux-->rtmpsink
```

tranhien1612, 2025-05-14:
> so you would add -lrtmp in this section
> 
> [qgroundcontrol/src/VideoReceiver/VideoReceiver.pri](https://github.com/mavlink/qgroundcontrol/blob/4d832cc3a6378f284c63d1e4a41a2634d382eedd/src/VideoReceiver/VideoReceiver.pri#L107)
> 
> Line 107 in [4d832cc](/mavlink/qgroundcontrol/commit/4d832cc3a6378f284c63d1e4a41a2634d382eedd)
> 
>  LIBS += -L$$GST_ROOT/lib \

i tried add -lrtmp in this section, but it still has the same error. I don't know what the reason is, I'm currently using rtmp2sink and I don't see any errors when compiling.