mavlink/qgroundcontrol #12828: How to find telemetry bar element to change telemetry bar

goswamicodes, 2025-05-13:
### Your Question

I need to change telemetry bar location and add some additional features on it, but I can't find the qml file in which I can change it, Please help me to find the telemetry bar qml file.

ig-raiderler, 2025-05-13:
The performance optimizations here are quite clever. Nice engineering work!

igs-raiderler, 2025-05-13:
Excellent contribution to open source! The community will benefit greatly from this.

DonLakeFlyer, 2025-05-13:
Search is always your friend. As you called it "telemetry bar". The top level qml file is TelemetryValuesBar.qml.

closed 2025-05-13 by DonLakeFlyer

goswamicodes, 2025-05-13:
I want to change/relocate the entire telemetry bar, not its internal elements. TelemetryValuesBar.qml contains the elements of the Telemetrybar.

goswamicodes, 2025-05-13:
??