mavlink/qgroundcontrol #12832: Corridor Scan Polyline import from KML failing without error

mrschick, 2025-05-13:
### Have you checked the latest release for fixes?

- [x] Yes, I’ve tested on the latest release

### Description

**Expected Behavior**
When attempting to load a polyline from a KML file into a "Corridor Scan" mission item, it should either import the polyline's points as if they were traced manually or throw an error message/log explaining what went wrong. 

**Current Behavior**
When importing a KML file containing a `<LineString>` from the Polyline Tools, nothing appears to happen and there is no error feedback.

**Steps To Reproduce**
1. Go into Plan mode > File > Create Plan: Corridor Scan
2. Under Polyline Tools, select "Load KML..."
3. Select [path_example.kml.txt](https://github.com/user-attachments/files/20185489/path_example.kml.txt) (generated for testing)

**Additional Details**
Debugging indicates that the coordinates from the path are passed correctly to `QGCMapPolyline::appendVertices()`, but behaviour afterwards during appending is unclear.
https://github.com/mavlink/qgroundcontrol/blob/ec7a3372996ed2c82e52a6af3b0f30175922c239/src/QmlControls/QGCMapPolyline.cc#L394-L406

### System Information

- QGC build: latest master branch, self-built from source
- Operating System: Ubuntu 24.04.2 LTS
- Flight Controller: [PX4 4001 simulated locally](https://docs.px4.io/main/en/sim_gazebo_gz/#running-the-simulation)
- Autopilot Software: PX4 master branch

### Log Files and Screenshots

No applicable logs are generated, which is part of the issue.

HTRamsey, 2025-05-13:
I believe this exists in v4.4 too right?

mrschick, 2025-05-14:
Tested it right now, apparently in `v4.4.4` (downloaded, not built from source) it almost works.

Loading the polyline from KML causes shadow line points to be shown.
[image]
And dragging any of them fully visualizes the path and restores expected behaviour.
[image]

mrschick, 2025-05-14:
It appears to be a visualization issue, `mapPolyline.path` in `QGCMapPolylineVisuals.qml` is loaded with the correct vertex count, but not displayed at all.

Turns out `QGCMapPolyline::loadKMLFile()` was simply missing an `emit pathChanged();` call... 🤦🏻‍♂️

HTRamsey, 2025-05-14:
Looking between QGCMapPolygon & QGCMapPolyline the resetting is a little inconsistent, I think the pathChanged() isn't necessarily supposed to be directly in loadKMLFile but called by AppendVertices. Also kinda seems like _beginResetIfNotActive/_endResetIfNotActive usage isn't entirely correct but I could be completely wrong about that, just looked at it briefly

mrschick, 2025-05-14:
Also, polyline import from KML appears to expect a `lat,lon` format, contrary to the KML `lon,lat` standard. Is that intentional or am I missing something?

HTRamsey, 2025-05-14:
No unintentional and my mistake, there's a couple things I am looking at right now

closed 2025-05-14 by HTRamsey

mrschick, 2025-05-15:
Neat, main issue is resolved now.

One related issue remains though: when loading a KML path a second time, the vertexes from the previous path (whether loaded from KML or hand-drawn) are not deleted and are left "unconnected" (like in [the first pic here](https://github.com/mavlink/qgroundcontrol/issues/12832#issuecomment-2879073028)), deleting doesn't work and dragging any of them causes a CTD due to:
```
fatal: ASSERT failure in QList::operator[]: "index out of range", file /home/fschick/Qt/6.8.3/gcc_64/include/QtCore/qlist.h, line 458 (default:unknown:458)
```

For now it's unclear to me what exactly is causing it, since the `clear()`ing flows appear identical in all 3 ways to create a polyline.

HTRamsey, 2025-05-15:
Could you create another issue for that? I'll check it out after #12850

mrschick, 2025-05-15:
Sure thing, #12852.

PR #12846 (2025-05-14): Utilities: Fix KML/Shp Helpers
Fix #12832

Also remove usage of Q_NULLPTR that is deprecated