mavlink/qgroundcontrol #12841: RTK settings page causing restart message to pop up

DonLakeFlyer, 2025-05-13:
### Have you checked the latest release for fixes?

- [x] Yes, I’ve tested on the latest release

### Description

* Connect RTK GPS to computer
* Boot QGC, set Clear settings on next boot
* Close, restart QGC
* Go to GPS indicator, expand
* Mystery "Restart application for changes to take affect" message appears. It's some from the survey in value I believe

### System Information

a

HTRamsey, 2025-05-13:
It's this guy:

[image]

DonLakeFlyer, 2025-05-14:
Yeah, that's kinda what I figured. Control problem with initialization most likely. That's going to cause problems in other places as well.

closed 2025-05-14 by DonLakeFlyer