mavlink/qgroundcontrol #12858: QGC crashes after disconnecting USB [Release V5.0]

diablo1281, 2025-05-15:
I can't add the tag - please mark this as `Release V5.0`.

**Clearly mention the version you're testing and your operating system.**
QGC https://github.com/mavlink/qgroundcontrol/commit/93ee76f13cb3054c5692751165e4b3fe704ade35
Windows 11 24H2

**Describe your hardware setup (telemetry, flight controller, firmware).**
Holybro Pixhawk 6X connected by USB cable.
Firmware: PX4 1.15.4

**Include detailed steps to reproduce the issue.**
After physically disconnecting the USB cable, while the connection in QGC is still open, I get an error message about access to the COM port.
Connection is still open (but with state `disconnected`), so I want to terminate it by clicking `Disconnect`. After clicking the button, the QGC window disappears, but the process continues working in the background. To launch QGC again, it needs to be manually closed via Task Manager as there can't be two instances running simultaneously. 

On the sidenote, it takes quite some time to launch every time.

**Add screenshots or screen recordings where possible — they really help!**

https://github.com/user-attachments/assets/286d0e7c-a8b9-45af-86fd-3c29c35a0f88

HTRamsey, 2025-05-15:
I'm gonna guess this is via autoconnect? Trying to open/close a composite USB gets stuck in a qt function for quite a while. If you manually set to the correct port I think this doesn't happen right?

diablo1281, 2025-05-15:
Just checked, it happens regardless if it's AutoConnect or set Link

hyeonseung89, 2025-05-16:
Even if you check with version v5.0rc, the same symptoms appear not only in USB serial connections but also in UDP and TCP connections.

Can you please refer to this part and check?

`critical: FactValueGrid::_initForNewVehicle: vehicle is NULL (default:FactValueGrid::_initForNewVehicle:66)`

HTRamsey, 2025-05-16:
@DonLakeFlyer looks like we may still have some issues after #12795

HTRamsey, 2025-05-16:
@diablo1281 @hyeonseung89 we'll have to wait for Don to look at it, but as far as I can tell there's nothing wrong with the comms code itself. The main thread just dies after _initForNewVehicle

HTRamsey, 2025-05-16:
@DonLakeFlyer nevermind I tracked it down to #12536 I think, the artifact from ec7a337 works but the one from  cd7128c does not. Although that PR doesn't look like it changes anything that would cause this... I will try to figure it out, @rubenp02 might need you to check it out too if you get a chance

PR #12864 (2025-05-16): Fix crash on usb disconnect, fix hiding of main window
* Fix #12858
* Fixes crash on usb cable disconnect
* Fixes main window disappearing on active vehicle going away
* Fixes link errors showing message in title and title in message area
* Stops popping link error dialog on usb cable pull
* Replacement for #12860
closed 2025-05-16 by DonLakeFlyer