mavlink/qgroundcontrol #12859: Update QGroundControl.com to use a more recent release for Mac

noah23olsen, 2025-05-16:
### Have you checked the latest release for fixes?

- [x] Yes, I’ve tested on the latest release

### Description

**Expected Behavior**
Describe what you thought should happen here.

The MacOSX download should work out of the box

**Current Behavior**
Describe what actually happened here.

It crashes. There was a [recent fix for this](https://github.com/mavlink/qgroundcontrol/pull/12614), but I had to download one of the [daily builds ](https://docs.qgroundcontrol.com/master/en/qgc-user-guide/releases/daily_builds.html)

The fix is to update the MacOSX download to a more recent release, or to add a link to download a daily build. 

**Steps To Reproduce**
1.Go to https://docs.qgroundcontrol.com/master/en/qgc-user-guide/getting_started/download_and_install.html
2. Download MacOSX version
3. It crashes instantly upon download.

**Additional Details**
Add any extra info (errors, warnings, etc.) here.

### System Information

Thanks

HTRamsey, 2025-05-16:
@julianoes could you update the release?

DonLakeFlyer, 2025-05-16:
Why bother with 5.0 already in RC mode?

HTRamsey, 2025-05-18:
So with v5.0 now having an rc, v4.4.4 macos won't likely be fixed

closed 2025-05-18 by HTRamsey

julianoes, 2025-05-18:
I asked @farhangnaderi for help to release v4.4.5 because I don't have a Mac, and I don't feel like figuring out GStreamer GitHub Action Magic, it's unpaid and the opposite of fun.

farhangnaderi, 2025-05-18:
> I asked @farhangnaderi for help to release v4.4.5 because I don't have a Mac, and I don't feel like figuring out GStreamer GitHub Action Magic, it's unpaid and the opposite of fun.

Should we still fix 4.4? I am confused!