mavlink/qgroundcontrol #12879: How to install QGroundControl on Apple M1 chip running Ubuntu in a VM?

MeethiKishori, 2025-05-19:
I would like to install QGroundControl (QGC) on an Apple M1 chip, running Ubuntu inside a virtual machine (VM). Are there any specific steps, dependencies, or workarounds required for this setup? 

If there are known issues or guides for ARM64 or Apple Silicon environments, please share. Any advice or pointers to relevant documentation would be appreciated.

mrpollo, 2025-05-19:
Please follow the docs, report if you find any issues

https://docs.qgroundcontrol.com/master/en/qgc-user-guide/getting_started/download_and_install.html

closed 2025-05-19 by mrpollo

MeethiKishori, 2025-05-19:
@mrpollo , But i need an ARM64-compatible build of QGroundControl. However, QGroundControl officially provides AppImages for x86_64 only, not ARM64.

DonLakeFlyer, 2025-05-19:
> However, QGroundControl officially provides AppImages for x86_64 only, not ARM64.

Correct. You'll have to build it yourself.

MeethiKishori, 2025-05-19:
i cant build, i dont know how to do.
So i thought to connect PX4 (inside Multipass) to QGroundControl (on macOS) via UDP.

i have the daily build of qgc, do you have suggestion on any specific version?

i did : make px4_sitl gz_x500 PX4_SIM_HOST_ADDR=192.168.0.74
output :  WARN  [health_and_arming_checks] Preflight Fail: No connection to the ground control station	
WARN  [health_and_arming_checks] Preflight Fail: No connection to the ground control station

DonLakeFlyer, 2025-05-20:
Just `make px4_sitl gz_x500`

Then read the PX4 docs on how to set UDP to broadcast mode for SITL. That's needed to cross the boundary between VM and regular OS. Depending on the VM manager you may need to do additional network settings but that varies.

MeethiKishori, 2025-05-20:
Thank you for advice.

Here is the step i followed:

From VM(multipass)

 make px4_sitl gz_x500
[0/1] cd /home/ubuntu/PX4-Autopilot/bu...topilot/build/px4_sitl_default/bin/px4

______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] startup script: /bin/sh etc/init.d-posix/rcS 0
INFO  [init] found model autostart file as SYS_AUTOSTART=4001
INFO  [param] selected parameter default file parameters.bson
INFO  [param] importing from 'parameters.bson'
INFO  [parameters] BSON document size 342 bytes, decoded 342 bytes (INT32:15, FLOAT:3)
INFO  [param] selected parameter backup file parameters_backup.bson
INFO  [dataman] data manager file './dataman' size is 1208528 bytes
INFO  [init] Gazebo simulator 8.9.0
INFO  [init] Starting gazebo with world: /home/ubuntu/PX4-Autopilot/Tools/simulation/gz/worlds/default.sdf
INFO  [init] Starting gz gui
INFO  [init] Waiting for Gazebo world...
INFO  [init] Gazebo world is ready
INFO  [init] Spawning model
INFO  [gz_bridge] world: default, model: x500_0
INFO  [lockstep_scheduler] setting initial absolute time to 3772000 us
INFO  [commander] LED: open /dev/led0 failed (22)
WARN  [health_and_arming_checks] Preflight Fail: ekf2 missing data
WARN  [health_and_arming_checks] Preflight Fail: system power unavailable
WARN  [health_and_arming_checks] Preflight Fail: No connection to the ground control station	
INFO  [uxrce_dds_client] init UDP agent IP:127.0.0.1, port:8888
INFO  [mavlink] mode: Normal, data rate: 4000000 B/s on udp port 18570 remote port 14550
INFO  [tone_alarm] home set
INFO  [mavlink] mode: Onboard, data rate: 4000000 B/s on udp port 14580 remote port 14540
INFO  [mavlink] mode: Onboard, data rate: 4000 B/s on udp port 14280 remote port 14030
INFO  [mavlink] mode: Gimbal, data rate: 400000 B/s on udp port 13030 remote port 13280
INFO  [logger] logger started (mode=all)
INFO  [logger] Start file log (type: full)
INFO  [logger] [logger] ./log/2025-05-20/20_45_19.ulg	
INFO  [logger] Opened full log file: ./log/2025-05-20/20_45_19.ulg
INFO  [mavlink] MAVLink only on localhost (set param MAV_{i}_BROADCAST = 1 to enable network)
INFO  [mavlink] MAVLink only on localhost (set param MAV_{i}_BROADCAST = 1 to enable network)
INFO  [px4] Startup script returned successfully
pxh> WARN  [health_and_arming_checks] Preflight Fail: No connection to the ground control station	
WARN  [health_and_arming_checks] Preflight Fail: No connection to the ground control station	
pxh> param MAV_0_BROADCAST = 1

param show MAV_0_BROADCAST
Symbols: x = used, + = saved, * = unsaved
x   MAV_0_BROADCAST [507,947] : 1

 1009/1925 parameters used.

"Can i make changes it in the file of px4, so that i dont have to set it up everytime it runs"

From MAC (qgc)

"I've configured UDP in QGC, but no data is appearing. How do I confirm if QGC is successfully receiving any UDP messages?"
Here is the screenshot of qgc

[image]

mrpollo, 2025-05-20:
This is a PX4 question, you should raise a question on their forums or github. 

https://discuss.px4.io