mavlink/qgroundcontrol #12881: Please offer compatibility with classic .params files, and ability to SAVE as classic file by default

AndKe, 2025-05-20:
### Feature Description

the QGC format for .params is not compatible with other GCS's , and hard to compare(diff)
readability is not helped by lines like this:

1	1	ATC_RAT_YAW_FLTD	0.000000000000000000	9
1	1	ATC_RAT_YAW_FLTE	1.000000000000000000	9
1	1	ATC_RAT_YAW_FLTT	20.000000000000000000	9
1	1	ATC_RAT_YAW_I	0.091200001537799835	9
1	1	ATC_RAT_YAW_IMAX	0.500000000000000000	9
1	1	ATC_RAT_YAW_P	0.912000000476837158	9
1	1	AUTO_OPTIONS	0	6

This many decimals is .. not needed and just bad for comparing/diff/version tracking
for exaxmple is clearly set to be: 
ATC_RAT_YAW_I	0.0912

### Flight Stacks

ArduPilot, PX4

### Vehicle Types

Multirotor, Fixed-wing, VTOL, Submarine

DonLakeFlyer, 2025-05-20:
So this was changed 9 years ago: https://github.com/mavlink/qgroundcontrol/issues/2951. The rational behind it which is to save in the file the actual value with full precision I think still makes sense.

AndKe, 2025-05-20:
The full precision is an illusion :) - as you see/know that the last digits were not actually entered, so in most cases the later decimals is just an error caused by the binary approximation due to limited mantissa bits of the fraction.  If we had actual interest in precision that far out we would store the values as decimal.Decimal - so I do understand the idea of storing "everything" - but there are no parameters that I can think of where this is useful.

There are additional reasons for supporting the classic .param (regardless of number of digits):

APMPlanner 2 and mavproxy can compare .param with actual settings /  this is useful for configuration check, or copying parts of the configuration between vehicles, or checking configuration against earlier "known good" state.
- for that one needs to have a compatible format.
It would be nice is QGC was one of the tools that could at least export to such file - and preferably read such files.

The simpler .param format is also the one "universal" format across GCS software.  It is odd that QGC is the only GCS that is not compatible with it. :)

DonLakeFlyer, 2025-05-20:
I thought it was just decimal places were the problem. Not sure where QGC's param format differs. Can you detail? 

Also FYI: QGC 5.0 supports showing you the diff between vehicle and param files when you load and allows you to selectively include/exclude.

DonLakeFlyer, 2025-05-21:
So I've been looking at the QGC format, which is: `Vehicle id, componentId, paramName, value, type`. It's been this way over 10 years, predates any of my work. That said there are advantages to this format:
* QGC save all parameters including params from vehicle components and could load them back into components
* Having the vehicle id allows you to warn about loading a file from the wrong vehicle

Adding the ability to save in the MP format makes sense, but I don't see that as the default since it would be a loss of functionality with respect to saving component parameters.

Also a bit too late in the game for 5.0, but I"ll tag most of this as 5.1. I will change the "Load File" wording to be more clear.

DonLakeFlyer, 2025-05-21:
@AndKe And thank you for always reporting problems and helping to make QGC better for so long now.