mavlink/qgroundcontrol #12886: Qt creator cannot recognize gstreamer.

vorsprungdurchtae, 2025-05-21:
### Your Question

Hi all,

I am currently developing QGC on Ubuntu Linux 20.04 and the following message keeps appearing

[cmake] CMake Error at /usr/local/share/cmake-3.29/Modules/FindPkgConfig.cmake:634 (message):
[cmake]   The following required packages were not found:
[cmake] 
[cmake]    - gstreamer-1.0>=1.20
[cmake] 
[cmake] Call Stack (most recent call first):
[cmake]   /usr/local/share/cmake-3.29/Modules/FindPkgConfig.cmake:862 (_pkg_check_modules_internal)
[cmake]   cmake/find-modules/FindGStreamer.cmake:300 (pkg_check_modules)
[cmake]   src/VideoManager/VideoReceiver/GStreamer/CMakeLists.txt:8 (find_package)
[cmake] 
[cmake] 
[cmake] 
[cmake] The command "/usr/local/bin/cmake -S /home/qgroundcontrol -B /home/qgroundcontrol/build" terminated with exit code 1.
[cmake] 
[cmake] Elapsed time: 00:01.

even though I already installed gstreamer 1.22.xx like 

~$ find /usr/local -name 'gstreamer-1.0.pc'
/usr/local/lib/x86_64-linux-gnu/pkgconfig/gstreamer-1.0.pc

~$ pkg-config --modversion gstreamer-1.0
1.22.12

Can anyone please burrow me idea?

Best,

HTRamsey, 2025-05-21:
[image]

vorsprungdurchtae, 2025-05-21:
Can you please be more precise? I don't really catch which file are u using.

Deepak-Beniya, 2025-05-27:
I am also facing the same issue.
How to solvbe it ?

HTRamsey, 2025-05-27:
Yeah if you have the correct GStreamer installed & in a standard location then FindGStreamer.cmake should find it. If it's not finding it that file has some variables you can set to help it.

closed 2025-07-03 by HTRamsey

foxalien, 2025-07-30:
You can check the dependencies on the https://docs.qgroundcontrol.com/master/en/qgc-dev-guide/getting_started/

Install Additional Packages (Platform Specific)

Ubuntu: sudo bash ./qgroundcontrol/tools/setup/install-dependencies-debian.sh

Fedora: sudo dnf install speech-dispatcher SDL2-devel SDL2 systemd-devel patchelf

Arch Linux: pacman -Sy speech-dispatcher patchelf

Mac sh qgroundcontrol/tools/setup/macos-dependencies.sh