mavlink/qgroundcontrol #12893: Deal with multiple composite ports on same device

DonLakeFlyer, 2025-05-21:
ArduPilot with a CubeOrange will expose two ports over a single usb connection. Only one of these is the mavlink connection. In the past, Stable 4.4 timeframe at least on Mac the first port was always the mavlink one. Hence Stable 4.4 would disregard the second port from the same device. That fix no longer works at least in OSX where it's the second port listed which is the mavlink connection.

Current 5.0 no longer does the secondary port connections and will try to connect to both ports. This works fine on Mac, but throws device errors on Linux on the non-mavlink port.

I thinking we are going to have to keep track of the composite ports associated with the same device. Connect to both, but don't show errors until we attempt the multiple connects. And if one of them connects we are good and eat the error on the other one. If both connect without errors and only one has mavlink then that is fine as well. Any maybe we then disconnect the other one.

DonLakeFlyer, 2025-05-21:
I think it's possible that although the list of ports returned by Qt might have the ports of out order that the ports are still named foo#1 and foo#2. Where foo#2 is the SLCAN port and is maybe always ends in a number larger then the first one which is the mavlink port.

dakejahl, 2025-05-21:
> Where foo#2 is the SLCAN port

Yeah I think this is the case. Anecdotally, I always select foo#2 when connecting for SLCAN on dronecan_gui_tool

DonLakeFlyer, 2025-05-21:
Also if you look at the driver definitions .inf file for both CubePilot and Holybro in the Mission Planner repo the are both set up this way as far as I can tell.

DonLakeFlyer, 2025-05-21:
@tridge Is there some way to detect which composite usb port set up by ArduPilot is the mavlink one without opening the port and waiting for mavlink on it? The hack we had before doesn't seem to be working anymore which was to just use the first one. In order to autoconnect QGC kinda needs to know.

DonLakeFlyer, 2025-05-21:
Although this comment "Second USB endpoint defaults to MAVLink (instead of SLCAN) except on CubePilot boards" I found in the ArduPilot source possibly shoots this whole thing in the head. Sigh...

tpwrules, 2025-05-21:
The functions are user configurable. On Linux the first (default mavlink) shows up suffixed with `-if00`. The second (various functions) shows up as `-if02`.

I don’t immediately know what Qt gives you on macOS but it is distinguishable over the wire.

DonLakeFlyer, 2025-05-22:
> On Linux the first (default mavlink) shows up suffixed with `-if00`.

Shows up from what tool/api? What value is that?

tpwrules, 2025-05-22:
It refers to the USB [interface descriptor](https://www.beyondlogic.org/usbnutshell/usb5.shtml#InterfaceDescriptors) number.

```
$ ls -l /dev/serial/by-id/
total 0
lrwxrwxrwx 1 root root 13 May 21 21:31 usb-Hex_ProfiCNC_CubeOrange_<hwserial>-if00 -> ../../ttyACM0
lrwxrwxrwx 1 root root 13 May 21 21:31 usb-Hex_ProfiCNC_CubeOrange_<hwserial>-if02 -> ../../ttyACM1
```

From `lsusb -vt`:
```
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/16p, 480M
    |__ Port 1: Dev 10, If 3, Class=CDC Data, Driver=cdc_acm, 12M
    |__ Port 1: Dev 10, If 1, Class=CDC Data, Driver=cdc_acm, 12M
    |__ Port 1: Dev 10, If 2, Class=Communications, Driver=cdc_acm, 12M
    |__ Port 1: Dev 10, If 0, Class=Communications, Driver=cdc_acm, 12M
```

The interface descriptor numbers are set by ArduPilot so they will be consistent. Interface 0 is parameter set `SERIAL0_` and interface 2 is (at least on Cubes, it varies by board) `SERIAL6_`. I noticed on my Mac that the Cube shows up as `/dev/tty.usbmodem101` and `/dev/tty.usbmodem103`. I would guess that the last digit refers to the (1-based instead of Linux) interface descriptor number. I don't know a lot about Windows but according to [this page](https://learn.microsoft.com/en-us/windows-hardware/drivers/install/standard-usb-identifiers#multiple-interface-usb-devices) you could expect to see `MI_00` and `MI_02` in the internal name somewhere (arbitrarily mapped to COM port number).

It's not clear to me without some more effort what Qt's interfaces give you on various platforms but maybe this is a useful clue?

DonLakeFlyer, 2025-05-22:
Thanks super helpful. I don't get that level of detail from Qt apis. But I might be able to use direct unix apis on Mac/Linux/Android and then something else on Windows.

tpwrules, 2025-05-22:
What is exposed by the various `QSerialPortInfo` methods? e.g. `description()`, `portName()`, `systemLocation()`...

I don't think there's a standard Unix API for this, but you might have to get the name with Qt then use a system-specific API to query that additional info.

DonLakeFlyer, 2025-05-22:
https://doc.qt.io/qt-6/qserialportinfo.html

DonLakeFlyer, 2025-05-22:
Nothing that helps with the identification. But I haven't logged systemLocation. Not sure if that hs anything special in there. I'll look tomorrow.

DonLakeFlyer, 2025-06-06:
So this is as good as it's gonna get for now

closed 2025-06-06 by DonLakeFlyer