mavlink/qgroundcontrol #12898: QGC 5 RC1 and RC2 can not be installed Android

Yannicflight, 2025-05-22:
### Your Question

I would love to help with testing QGCS V5 on Android but for some reason both RC1 and RC2 cannot  be installed on my Redmi TAB pro. 
It just says the package is not correct when I try to install the APK. 

There is nog 32 or 64bit versions anymore of the apk so I assume V5 is 64bit only?

HTRamsey, 2025-05-23:
It's the same APK for armeabi-v7a & arm64-v8a now

Yannicflight, 2025-05-23:
I kept getting the  "app not installed as package appears to be invalid" error.
Tried installing trough adb and got an error that the app was unsigned.

After signing it using uber-apk-signer i could install the app trough adb.
Will test it now in the coming days.

For others who want to test on Android, the signed version of the RC2 apk can be found here:
https://drive.google.com/file/d/1o-EUuIbijEZOd8LqXbOyujEMhoAiNCGf/view?usp=sharing

HTRamsey, 2025-05-23:
Where are you downloading them? They should be signed already

Yannicflight, 2025-05-23:
Directly from the github assets under RC2

HTRamsey, 2025-05-23:
Oh weird, you can grab a download from the Continuous Release which is close enough and should work because I just checked it

PR #12907 (2025-05-24): CI: Always Sign Android if Possible
Closes #12819
Closes #12898
We can always sign if the github secrets are accessible