mavlink/qgroundcontrol #12938: Latest APK will not install on Android

joe-stroup, 2025-05-28:
### Have you checked the latest release for fixes?

- [x] Yes, I’ve tested on the latest release

### Description

After downloading the latest release of QGC, it will not install.  It presents this message:

[image]

### System Information

This is the latest version on the QGC website.  I get the same results on two android devices, a Google Pixel 4XL and a Samsung Galaxy A4 Tab tablet.

HTRamsey, 2025-05-28:
I think the APKs weren't being signed for the RC which is fixed and will be in the next release, but for now you can try the daily build instead if you want which I know is signed correctly

HTRamsey, 2025-05-30:
I have checked and RC3 is signed now

closed 2025-05-30 by HTRamsey